lancaster-university/microbit-dal #339: Bluetooth UART doesn't work

Lechucico, 2018-02-06:
I have the following code:

```
#include "MicroBit.h"
#include "MicroBitUARTService.h"

MicroBitUARTService *uart;
MicroBit uBit;

uint8_t connected = 0;

void onConnect(MicroBitEvent) {
    connected = 1;
    uBit.display.print("C");
    uBit.serial.printf("Connected");
}

void onDisconnect(MicroBitEvent) {
    connected = 0;
    uBit.display.print("D");
    uBit.serial.printf("Disconnected");
}

void onButtonA(MicroBitEvent e) {
    if (connected == 0) {
        uBit.display.print("X");
        return;
    }
    int x = uBit.accelerometer.getX();
    int y = uBit.accelerometer.getY();
    int z = uBit.accelerometer.getZ();
    uBit.display.print("A");
    uBit.serial.printf("Button: A \r\n x:%i y:%i z:%i \r\n",x,y,z);
    uart->send(ManagedString("Button A"));
}

void onButtonB(MicroBitEvent e) {
    if (connected == 0) {
        uBit.display.print("X");
        return;
    }
    int x = uBit.accelerometer.getX();
    int y = uBit.accelerometer.getY();
    int z = uBit.accelerometer.getZ();
    uBit.display.print("B");
    uBit.serial.printf("Button: B \r\n x:%i y:%i z:%i \r\n",x,y,z);
    uart->send(ManagedString("Button B"));
}

int main() {
    // Initialise the micro:bit runtime.
    uBit.init();

    uBit.messageBus.listen(MICROBIT_ID_BLE, MICROBIT_BLE_EVT_CONNECTED, onConnect);
    uBit.messageBus.listen(MICROBIT_ID_BLE, MICROBIT_BLE_EVT_DISCONNECTED, onDisconnect);
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_A, MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_B, MICROBIT_BUTTON_EVT_CLICK, onButtonB);

    uart = new MicroBitUARTService(*uBit.ble, 32, 32);

    uBit.display.print("S");
    uBit.serial.printf("Starting Micro:Bit Program \r\n");

    release_fiber();
}

```

I pair my micro:bit with my macbook. The problem is that I don't know how to read this uart information sent over bluetooth.

I did not find any example for reading UART  on a pc/macbook/raspberry.

jaustin, 2018-02-15:
The micro:bit UART is based on the Nordic UART, which can be used with a number of existing mobile apps
https://support.microbit.org/support/solutions/articles/19000062330-using-the-micro-bit-bluetooth-low-energy-uart-serial-over-bluetooth-

But you can also use native code to talk to the uart. One such example is this
https://github.com/tigoe/BluetoothLE-Examples/tree/master/noble/readSerial

I haven't used the UART myself on anything except mobile, but the knowledgebase article listed above was written using the MakeCode wrapper for the DAL's UART service and the apps listed at the bottom of the page.

So to be clear - are you saying you think that the DAL UART is broken, or just that you're not sure how to communicate with it on your desktop/laptop?

Lechucico, 2018-02-15:
I'm just saying that i'm not sure how to communicate using my laptop. Actually I've made an android application but this doesn't work. Can you check this and tell me what I'm doing bad?

```
public class MainActivity extends Activity {

    private BluetoothAdapter bleAdapter;
    private BluetoothGattCharacteristic UART;
    private BluetoothGattDescriptor descriptor;
    private TextView connectionState;

    private final UUID SERVICE = UUID.fromString("6E400001-B5A3-F393-E0A9-E50E24DCCA9E");
    private final UUID CHARACTERISTIC = UUID.fromString("6E400003-B5A3-F393-E0A9-E50E24DCCA9E");
    private final UUID CONFIG_DESCRIPTOR = UUID.fromString("00002902-0000-1000-8000-00805f9b34fb");

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        if (!getPackageManager().hasSystemFeature(PackageManager.FEATURE_BLUETOOTH_LE)) {
            Toast.makeText(this, "BLE Not Supported", Toast.LENGTH_SHORT).show();
            finish();
        }

        bleAdapter = ((BluetoothManager) getSystemService(BLUETOOTH_SERVICE)).getAdapter();

        if (bleAdapter == null || !bleAdapter.isEnabled()) {
            Intent enableBtIntent = new Intent(BluetoothAdapter.ACTION_REQUEST_ENABLE);
            startActivityForResult(enableBtIntent, 1);
        }

        Set<BluetoothDevice> pairedDevices = bleAdapter.getBondedDevices();

        for (BluetoothDevice device : pairedDevices) {

            device.connectGatt(this, true, new BluetoothGattCallback() {

                @Override
                public void onConnectionStateChange(BluetoothGatt gatt, int status, int newState) {
                    super.onConnectionStateChange(gatt, status, newState);

                    switch (newState) {
                        case BluetoothProfile.STATE_CONNECTED:
                            //UART = gatt.getService(SERVICE).getCharacteristic(CHARACTERISTIC);
                            //gatt.setCharacteristicNotification(UART,true);
                            //descriptor = UART.getDescriptor(CONFIG_DESCRIPTOR);
                            //descriptor.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
                            //gatt.writeDescriptor(descriptor);

                            runOnUiThread(new Runnable() {
                                public void run() {
                                    connectionState = findViewById(R.id.state);
                                    connectionState.setText("Connected = True");
                                    connectionState.setTextColor(Color.GREEN);
                                }
                            });
                            break;

                        case BluetoothProfile.STATE_DISCONNECTED:
                            //UART = gatt.getService(SERVICE).getCharacteristic(CHARACTERISTIC);
                            //gatt.setCharacteristicNotification(UART,false);
                            //descriptor = UART.getDescriptor(CONFIG_DESCRIPTOR);
                            //descriptor.setValue(BluetoothGattDescriptor.DISABLE_NOTIFICATION_VALUE);
                            //gatt.writeDescriptor(descriptor);

                            runOnUiThread(new Runnable() {
                                public void run() {
                                    connectionState = findViewById(R.id.state);
                                    connectionState.setText("Connected = False");
                                    connectionState.setTextColor(Color.RED);
                                }
                            });
                    }
                }

                @Override
                public void onCharacteristicChanged(BluetoothGatt gatt, BluetoothGattCharacteristic characteristic) {
                    super.onCharacteristicChanged(gatt, characteristic);

                    TextView x = (TextView) findViewById(R.id.x_axis);
                    TextView y = (TextView) findViewById(R.id.y_axis);
                    TextView z = (TextView) findViewById(R.id.z_axis);

                    x.setText(characteristic.getValue().toString());
                    y.setText(characteristic.getValue().toString());
                    z.setText(characteristic.getValue().toString());
                }
            });
        }
    }
}

```

closed 2018-02-27 by Lechucico