lancaster-university/microbit-dal #342: How to create custom messageBus event

Lechucico, 2018-02-27:
The events from the accelerometers are not useful for me, because I need to raise an event when a movement is performed (event if it just moves a little bit the micro:bit). 

How I can create a new handler that checks the accelerometer data and raises an event when it detects movement?

Example:

```
#include "MicroBit.h"
#include "MicroBitUARTService.h"
MicroBitUARTService *uart;
MicroBit uBit;

void onAwake (MicroBitEvent e){
   uart->send(ManagedString("Movement Detected));
}

int main (){
   uBit.init();
   uBit.messageBus.listen("MICROBIT_ID_CUSTOM", "MICROBIT_EVT_CUSTOM", onAwake);
   uart = new MicroBitUARTService(*uBit.ble, 32, 32);
   release_fiber();
}
```

bluetooth-mdw, 2018-02-27:
Bitty Software tutorials include code showing how to create events and use the event service. See http://bittysoftware.com/tutorials/bitty_controller_4tronix_yotta.html

To poll the accelerometer see https://lancaster-university.github.io/microbit-docs/ubit/accelerometer/

Lechucico, 2018-02-27:
I think the best approach is is on main fiber read the accelerometer data each 100ms, compare the last acceleration with the current acceleration and generate an event if it passes some kind of threshold. Is that the best approach for raising an event for an accelerometer change?

bluetooth-mdw, 2018-02-27:
Sounds good to me :-)

Lechucico, 2018-02-28:
This is the way to do it?

```
const int ACCELEROMETER_EVENT = 9020;

void onAwake (MicroBitEvent e, int x, int y, int z){
     // some code
}

int main() {
    uBit.init();
    uBit.messageBus.listen(ACCELEROMETER_EVENT, 0, onAwake);

    while (1) {
        int x = uBit.accelerometer.getX();
        int y = uBit.accelerometer.getY();
        int z = uBit.accelerometer.getZ();
        MicroBitEvent evt(ACCELEROMETER_EVENT, x, y, z);
        uBit.sleep(100);
    }
}
```

bluetooth-mdw, 2018-03-01:
No, you cannot construct a MicroBitEvent object like that. Look at https://lancaster-university.github.io/microbit-docs/data-types/event/ and you'll see the constructor takes a 16-bit ID (which you have right) and a 16-bit value (which you have not).

If you need all three of X, Y and Z then you will probably need to send them in separate event objects but look at https://lancaster-university.github.io/microbit-docs/ubit/accelerometer/ and see if there are other ways of achieving what you want. Could the predefined gestures help, for example?

Lechucico, 2018-03-01:
enfoqueI've seen the constructor, but I was thinking that there has to be an option to send data to that event generation. 

I need to raise events when small movements occurs on Micro:Bit. All the default gestures from accelerometer need a lot of movement to activate. That's why I check the x,y,z on main, and raise an event if the accelerometer has some small movement.

How I can send that information in separate event objects?

In any case, I can just call a method instead of creating an event, I've just wanted this approach for performing the event task asyncronous, with a method call it will have to wait until that method finishes before going to sleep.

bluetooth-mdw, 2018-03-01:
Yes, you can send X, Y and Z values in separate event objects, just give them different event IDs. I do something similar with sensor data here: http://bittysoftware.com/tutorials/solutions/controller/bc_4tronix.cpp 

Look at sampleSensors and transmitSensorValue

Events get fired in their own fiber so you probably don't need to worry about fibers.

Lechucico, 2018-03-01:
@bluetooth-mdw Thanks, I'll take a look.

Another question: The values of my accelerometer when on my table are the following: x=0, y=0, z=1050. This is gravity? Because I'm trying to detect movement and the z axis is always a huge value.

bluetooth-mdw, 2018-03-01:
Yes. All of this information is in the documentation, by the way: "The force measured in the Z axis, in milli-g" from https://lancaster-university.github.io/microbit-docs/ubit/accelerometer/#getz

(hint)

Lechucico, 2018-03-02:
@bluetooth-mdw there is anyway of removing the gravity force from a@bluetooth-mdw there is a way of removing the gravity force from the accelerometer readings?

bluetooth-mdw, 2018-03-02:
Yes, subtraction. 1g = 1000milli-g.

Lechucico, 2018-03-02:
Just substract 1000mili-g? So, when calculating acceleration this will be: sqrt((x*x+y*y+z*z)-1000)?

bluetooth-mdw, 2018-03-02:
No. I thought you meant you wanted the device at rest to deliver a Z value of 0. So I meant subtract 1000 from the Z reading. More than that, I cannot say.