lancaster-university/microbit-dal #351: Animate function does not work with negative stride

microbit-mark, 2018-04-05:
When stride argument for the animatemethod has a neagtive value the micro:bit does nothing. We would expect it to animate in the opposite direction.
eg
```cpp
#include "MicroBit.h"

MicroBit uBit;

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    // Setup a simple triangular waveform.
    MicroBitImage img("1 0 0 0 0 0 0 0 0 1\n0 1 0 0 0 0 0 0 1 0\n0 0 1 0 0 0 0 1 0 0\n0 0 0 1 0 0 1 0 0 0\n0 0 0 0 1 1 0 0 0 0\n");

     uBit.display.animate(img, 200, -1);
}
```

carlosperate, 2018-04-05:
Just to clarify @microbit-mark, when this code is run on a micro:bit nothing is displayed in the LED matrix, is that correct?

microbit-mark, 2018-04-05:
Yep. nothing is displayed on the micro:bit. Using `uBit.display.animate(img, 200, 1);` works

finneyj, 2018-04-05:
Thanks @microbit-mark  and @carlosperate 

Based on a quick scan of the code, it looks like you currently need to specify an explicit starting position if you want negative scrolling: https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitDisplay.cpp#L934

Can you try your code again with the optional fourth parameter set, as indicated here:
https://lancaster-university.github.io/microbit-docs/ubit/display/#int-animate-microbitimage-image-int-delay-int-stride-int-startingposition

I guess it will be roughly ```-the_width_of_your_image``` ...

microbit-mark, 2018-04-10:
I've tried variations on `uBit.display.animate(img, 200, -1,-10)` but can't get a working example of negative scrolling animation using the waveform sample.

microbit-mark, 2018-04-24:
@finneyj anything else I can try?

microbit-pauline, 2018-11-06:
Assigning to @microbit-sam as a BAU DAL activity

microbit-mark, 2019-02-01:
Had another request in support for this, so polling the issue.

microbit-mark, 2019-03-05:
micro:bit support: 20935
Another support request for this issue

martinwork, 2019-03-05:
I think I have fixed this. I'll make a PR.

microbit-mark, 2020-06-30:
micro:bit support: 38373
Another support request re this issue
Are we able to merge the fix?

microbit-mark, 2020-12-17:
Raised again in  microbit support: 42107

microbit-mark, 2021-04-16:
Raised in microbit support: 44708 Also reproduces on V2. @martinwork does your PR https://github.com/lancaster-university/microbit-dal/pull/427 still apply for both board variants? This would be a good one to progress as it crops up every so often.

martinwork, 2021-04-16:
@microbit-mark V2 and V1 look the same with negative stride, using the test below, so hopefully the changes needed in CODAL would be similar. Would you like a PR like this for the DAL js-event-semantics branch and one for CODAL?

```
#include "MicroBit.h"

MicroBit uBit;

MicroBitImage img("255 0 0 0 0 0 0 0 0 255\n"
                  "0 255 0 0 0 0 0 0 255 0\n"
                  "0 0 255 0 0 0 0 255 0 0\n"
                  "0 0 0 255 0 0 255 0 0 0\n"
                  "0 0 0 0 255 255 0 0 0 0\n");

void onButtonA(MicroBitEvent e)
{
    int stride = -1;
    uBit.display.stopAnimation();
    uBit.display.clear();
    uBit.display.animate(img, 1000, stride, 2, 0);
}

void onButtonB(MicroBitEvent e)
{
    int stride = 1;
    uBit.display.stopAnimation();
    uBit.display.clear();
    uBit.display.animate(img, 1000, stride, 2, 0);
}

int  main()
{
    uBit.init();

    uBit.messageBus.listen( MICROBIT_ID_BUTTON_A,  MICROBIT_BUTTON_EVT_CLICK, onButtonA);
    uBit.messageBus.listen( MICROBIT_ID_BUTTON_B,  MICROBIT_BUTTON_EVT_CLICK, onButtonB);

    release_fiber();
}

```

microbit-mark, 2021-07-08:
Raised again in micro:bit support: 46440 

@martinwork 
> Would you like a PR like this for the DAL js-event-semantics branch and one for CODAL?

It would be great to fix this for all board variants . I'll leave the prioritisation with you and @finneyj as to where and when.

martinwork, 2022-04-06:
DAL  js-event-semantics branch: https://github.com/lancaster-university/microbit-dal/pull/485
CODAL: https://github.com/lancaster-university/codal-core/pull/145

pacorreia, 2024-11-16:
Wow, seems this went silent, since 2022, I started using this to teach code to my son, and while showing him cool things to do with led matrix, I remembered to show him the Knight Rider light effect, and surprise, although the MakeCode emulator works, when it compiles and sends it to the board it doesn't work.

It's a V2 board, any news about a fix? is there any PR waiting?