lancaster-university/microbit-dal #355: Bluetooth notifications transmission times very irregular sometimes

bluetooth-mdw, 2018-05-11:
I've had a few people report issues (usually via the microbit foundation support team) with the frequency with which they receive accelerometer data over Bluetooth. This is configurable via the Accelerometer Period characteristic. 

With a period of 640ms it works fine. At 20ms though, notifications arrive with a frequency that doesn't seem related to this value at all. Testing with nRF Connect :

```
I 12:07:47.651 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-20-01
A 12:07:47.651 "(0x) 00-01-A0-03-20-01" received
I 12:07:47.651 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-10-01
A 12:07:47.651 "(0x) 00-01-A0-03-10-01" received
I 12:07:47.651 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-10-01
A 12:07:47.651 "(0x) 00-01-A0-03-10-01" received
I 12:07:47.681 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-10-01
A 12:07:47.681 "(0x) 00-01-A0-03-10-01" received
I 12:07:47.682 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-10-01
A 12:07:47.682 "(0x) 00-01-A0-03-10-01" received
I 12:07:47.726 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-10-01
A 12:07:47.726 "(0x) 00-01-A0-03-10-01" received
I 12:07:47.727 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 10-01-A0-03-10-01
A 12:07:47.727 "(0x) 10-01-A0-03-10-01" received
I 12:07:47.772 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-10-01
A 12:07:47.772 "(0x) 00-01-A0-03-10-01" received
I 12:07:47.772 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 10-01-A0-03-10-01
A 12:07:47.772 "(0x) 10-01-A0-03-10-01" received
I 12:07:47.772 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-10-01
A 12:07:47.772 "(0x) 00-01-A0-03-10-01" received
I 12:07:47.817 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 00-01-A0-03-10-01
A 12:07:47.817 "(0x) 00-01-A0-03-10-01" received
I 12:07:47.817 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 10-01-A0-03-10-01
A 12:07:47.817 "(0x) 10-01-A0-03-10-01" received
```
Looks like notifications are being buffered and transmitted in batches, maybe.

whereas at 640ms periodicity:

```
I 12:24:31.511 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 10-00-D0-03-C0-00
A 12:24:31.511 "(0x) 10-00-D0-03-C0-00" received
I 12:24:32.189 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 10-00-D0-03-C0-00
A 12:24:32.189 "(0x) 10-00-D0-03-C0-00" received
I 12:24:32.768 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 30-00-E0-03-D0-00
A 12:24:32.768 "(0x) 30-00-E0-03-D0-00" received
I 12:24:33.398 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 60-00-D0-03-E0-00
A 12:24:33.398 "(0x) 60-00-D0-03-E0-00" received
I 12:24:33.981 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 80-00-D0-03-20-01
A 12:24:33.981 "(0x) 80-00-D0-03-20-01" received
I 12:24:34.618 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) F0-FF-20-03-D0-01
A 12:24:34.618 "(0x) F0-FF-20-03-D0-01" received
I 12:24:35.250 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 50-FF-E0-02-C0-02
A 12:24:35.250 "(0x) 50-FF-E0-02-C0-02" received
I 12:24:35.879 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 40-FF-D0-02-E0-02
A 12:24:35.879 "(0x) 40-FF-D0-02-E0-02" received
I 12:24:36.455 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 50-FF-F0-02-B0-02
A 12:24:36.455 "(0x) 50-FF-F0-02-B0-02" received
I 12:24:37.095 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 50-FF-D0-02-A0-02
A 12:24:37.095 "(0x) 50-FF-D0-02-A0-02" received
I 12:24:37.720 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 90-FF-E0-02-F0-01
A 12:24:37.720 "(0x) 90-FF-E0-02-F0-01" received
I 12:24:38.354 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 70-FF-F0-02-A0-02
A 12:24:38.354 "(0x) 70-FF-F0-02-A0-02" received
I 12:24:38.937 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) 60-FF-10-03-60-02
A 12:24:38.937 "(0x) 60-FF-10-03-60-02" received
I 12:24:39.569 Notification received from e95dca4b-251d-470a-a062-fa1922dfa9a8, value: (0x) C0-FF-80-03-D0-01
A 12:24:39.569 "(0x) C0-FF-80-03-D0-01" received
```

This behaviour messes up some of the experiments people are doing with micro:bits moving in space, in various scenarios. It would be greast if it were possible to smooth the timing of the transmission of notifications so this was more aligned with the timing of the production of the data being notified.

cpseager, 2018-05-11:
Not just accelerometer - none of the sensors transmit regularly/reliably at fast speeds over BLE e.g. see #316. In my experience lots of the expected notifications are just lost completely.

bluetooth-mdw, 2018-05-11:
Wondering if there needs to be some flow control to stop overwhelming the nRF stack? I don't have time to look into this more unfortunately.