lancaster-university/microbit-dal #356: Query: ability to alter ADC ref and prescaler in Javascript or Javascript blocks on Micro:bit?

DrSianArmstrong, 2018-05-15:
I can see how to do this in Micropython, and I assume from what I have read that unless set otherwise the reference is the supply and the prescaler is set to 1:1, but is there a way within Javascript of Javascript blocks of altering these parameters for ADC of voltages on the input pins?
Thanks

(P.S. New to posting issues on github, so apologies if I have broken any ettiquette...)

jaustin, 2018-06-29:
Hi @DrSianArmstrong - thanks for filing the issue here. 

There isn't yet a way to configure the ADC behaviour from Javascript, sadly - the DAL doesn't expose any API for that but we have an issue #322 to track this request. The issue is that we'd need to extend mbed to allow us to modify the parameters for the prescaler and reference.

The configuration of the ADC is here:
https://github.com/lancaster-university/mbed-classic/blob/microbit_hfclk%2Bmb6/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/analogin_api.c#L53
(IE we're using standard mbed configuration of supply voltage and 1/3 prescaling)

Jonny

whaleygeek, 2019-05-06:
@DrSianArmstrong There is a MicroPython gist here from @microbit-carlos showing how to do this, but you probably know that already based on what you say in your question.

https://gist.github.com/microbit-carlos/baebc389bccc4366abf3701acc7fc61c

I'm not aware of any Javascript API's for writing direct to registers though, but you could do it with a MakeCode package and a small CPP file. I understand that if you provide some CPP in your package, Microsoft automatically marshall your CPP file to a compiler on their Azure server on first use, so you shouldn't need to install all the compiler tools locally.

You can find info about writing a MakeCode package here:

https://makecode.com/extensions/getting-started

Note, you don't need to publish the package in order to use it, just paste it's github url into the ADD PACKAGE search page and it will import it for you.

pelikhan, 2019-05-06:
If you look at the Microsoft/pxt-ws2812b repo you can see how to ingetrage ASM into your project.