lancaster-university/microbit-dal #365: Start / End Addresses in Memory Map incorrect

microbit-sam, 2018-08-01:
The address stored in the memory map don't make sense!

```
MicroBitPartialFlashingService calculates the beginning of the MakeCode region by rounding FLASH_PROGRAM_END:

without BLE  with BLE
FLASH_PROGRAM_END 0x00031ac8   0x000333d4
MakeCode start    0x00031c00   0x00033400
MakeCode magic    0x00031c00   0x00033800
```

martinwork, 2018-08-22:
How about the following? It stores the address where the magic is found in the MakeCode region start address.

`
/*
 * Function to fetch the hashes from a PXT generated build
 */
void MicroBitMemoryMap::findHashes()
{
    uint8_t magic[16] = { 0x70, 0x8E, 0x3B, 0x92, 0xC6, 0x15, 0xA8, 0x41, 0xC4, 0x98, 0x66, 0xC9, 0x75, 0xEE, 0x51, 0x97 };

    // Iterate through pages to find magic
    uint32_t add = FLASH_PROGRAM_END;
    uint32_t rem = add % PAGE_SIZE;
    if ( rem) add += PAGE_SIZE - rem;
    for( ; add < DEFAULT_SCRATCH_PAGE; add += PAGE_SIZE)
    {
        uint32_t *magicAddress = (uint32_t *)add;

        // Check for 16 byte Magic
        if ( memcmp( magicAddress, magic, 16)) {
                // If the magic has been found use the address and the hashes that follow
                memoryMapStore.memoryMap[2].startAddress = magicAddress;
                memcpy( memoryMapStore.memoryMap[1].hash, magicAddress + 4, 8);
                memcpy( memoryMapStore.memoryMap[2].hash, magicAddress + 6, 8);
                return;
        }
    }
}

`

microbit-sam, 2018-08-22:
Yep this looks like the right idea - thanks Martin!

I'll have a go at implementing this, maybe with the mumur hash instead? How fast could you hash the DAL when you used it? Originally chose to use the embedded hashes to avoid having to hash the flash on the device

martinwork, 2018-08-22:
The thing about the code above is that it provides the correct makecode region start address rather than the one calculated by rounding FLASH_PROGRAM_END.

Calculating the hash is the way we need to go. Is there a need to check the embedded one as well? The suggestion was that we check the embedded hashes before calculating one.

finneyj, 2018-09-10:
@microbit-sam @martinwork 

Looks like we can safely close this issue off now?

microbit-sam, 2018-09-10:
Yep, Martin's fixed these for me :)

closed 2018-09-10 by microbit-sam