lancaster-university/microbit-dal #377: (BLE) Ability to remove default GATT services by code

orklann, 2018-08-25:
Currently, I make a custom GATT services, and It just works only by removing some default GATT services in `pxt.json` like this:

```
"yotta": {
        "config": {
            "microbit-dal": {
                "bluetooth": {
                    "dfu_service": 0,
                    "event_service": 0,
                    "device_info_service": 0,
                    "security_level": "SECURITY_MODE_ENCRYPTION_NO_MITM"
                },
                "gatt_table_size": "0x700"
            }
        }
    }
```
In the above config, it removes `dfu_service`, `event_service`, `device_info_service`, it works fine.
But after I publish my package, it would better with zero config for the users to just use my package. 
So I wonder if I can remove these 3 service in my package with code (in C++)?

jaustin, 2018-08-28:
@orklann another option would be to leave these as compile time options but make them configurable in MakeCode in the same way that security_level is... Would that work well for you?

From a micro:bit point of view, turning these off makes it hard to guarantee that the apps will work smoothly (we currently tell people you don't need to enter 'Pairing/Bluetooth' mode to connect to your device if you have the Bluetooth package included, but this would change that)

Is this a way to save RAM, or something else?

orklann, 2018-08-28:
@jaustin yes, it also works by making the same configuration in MakeCode.

```
Is this a way to save RAM, or something else?
```

My custom GATT service was coded in C++, and exposed to MakeCode via Typescript, actually I don't know why It does not work without turning off some default GATT services, and so it's not about saying RAM or something else.

I think I should make some notices to the users of my package to turned off some default GATT services via the configuration files in MakeCode, thanks for your suggestion.

pelikhan, 2018-10-08:
Unfortunately most BLE options are C++ precompiler options and cannot be changed at runtime. 

Note there's more discussion about controlling BLE memory at https://github.com/lancaster-university/microbit-dal/issues/390