lancaster-university/microbit-dal #384: MicroBitMagnetometerService gets stuck on dal-integration-4

microbit-sam, 2018-09-08:
Program gets stuck starting the service!

```
(gdb) bt
#0  0x00022a5c in tmr1_getCounter64 () at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/us_ticker.c:156
#1  0x00022b6a in us_ticker_read () at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/mbed-classic/targets/hal/TARGET_NORDIC/TARGET_MCU_NRF51822/us_ticker.c:206
#2  0x0002b9ca in wait_us (us=us@entry=100000) at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/mbed-classic/common/wait_api.c:29
#3  0x0002b9de in wait_ms (ms=ms@entry=100) at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/mbed-classic/common/wait_api.c:24
#4  0x0001ac6e in MicroBitCompassCalibrator::calibrateUX (this=0x20002374 <uBit+464>)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitCompassCalibrator.cpp:402
#5  0x0002899c in MemberFunctionCallback::methodCall<MicroBitCompassCalibrator> (object=<optimized out>, method=<optimized out>, e=...)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/inc/core/MemberFunctionCallback.h:111
#6  0x0002a4c8 in MemberFunctionCallback::fire (this=this@entry=0x20002bc8, e=...)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/core/MemberFunctionCallback.cpp:47
#7  0x00029256 in async_callback (param=0x20002b9c) at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitMessageBus.cpp:113
#8  0x0001b86e in MicroBitMessageBus::process (this=this@entry=0x20002274 <uBit+208>, evt=..., urgent=urgent@entry=true)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitMessageBus.cpp:369
#9  0x0002931c in MicroBitMessageBus::queueEvent (this=0x20002274 <uBit+208>, evt=...)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitMessageBus.cpp:157
#10 0x00029382 in MicroBitMessageBus::send (this=<optimized out>, evt=...)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitMessageBus.cpp:324
#11 0x00019e7e in MicroBitEvent::fire (this=this@entry=0x20003e88) at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/types/MicroBitEvent.cpp:83
#12 0x0002840c in MicroBitEvent::MicroBitEvent (this=0x20003e88, source=<optimized out>, value=<optimized out>, mode=<optimized out>)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/types/MicroBitEvent.cpp:64
#13 0x0001a8fc in MicroBitCompass::calibrate (this=this@entry=0x20002b38)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitCompass.cpp:225
#14 0x0001aa9a in MicroBitCompass::heading (this=0x20002b38) at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitCompass.cpp:169
#15 0x00027f98 in MicroBit::onListenerRegisteredEvent (this=0x200021a4 <uBit>, evt=...)
---Type <return> to continue, or q <return> to quit---
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit/source/MicroBit.cpp:220
#16 0x00027f2c in MemberFunctionCallback::methodCall<MicroBit> (object=<optimized out>, method=<optimized out>, e=...)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/inc/core/MemberFunctionCallback.h:111
#17 0x0002a4c8 in MemberFunctionCallback::fire (this=this@entry=0x20002db8, e=...)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/core/MemberFunctionCallback.cpp:47
#18 0x00029256 in async_callback (param=0x20002d8c) at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitMessageBus.cpp:113
#19 0x00019ab8 in invoke (entry_fn=entry_fn@entry=0x29205 <async_callback(void*)>, param=param@entry=0x20002d8c)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/core/MicroBitFiber.cpp:569
#20 0x0001b882 in MicroBitMessageBus::process (this=this@entry=0x20002274 <uBit+208>, evt=..., urgent=urgent@entry=false)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitMessageBus.cpp:371
#21 0x000293a8 in MicroBitMessageBus::idleTick (this=0x20002274 <uBit+208>)
    at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/drivers/MicroBitMessageBus.cpp:281
#22 0x00019868 in idle () at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/core/MicroBitFiber.cpp:925
#23 0x000198e8 in schedule () at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/core/MicroBitFiber.cpp:837
#24 0x00019996 in fiber_sleep (t=t@entry=5) at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit-dal/source/core/MicroBitFiber.cpp:363
#25 0x00018252 in MicroBit::sleep (this=0x200021a4 <uBit>, milliseconds=5) at /Users/admin/Projects/microbit/OutOfBoxExperience/yotta_modules/microbit/inc/MicroBit.h:457
#26 wake () at /Users/admin/Projects/microbit/OutOfBoxExperience/source/OutOfBox.cpp:141
#27 0x00018e7a in main () at /Users/admin/Projects/microbit/OutOfBoxExperience/source/OutOfBox.cpp:627
```

finneyj, 2018-09-08:
Hmm. Is this repeatable / deterministic?

microbit-sam, 2018-09-08:
Calibration runs but nothing on screen!
If you manage to complete the invisible calibration the program continues to run

```
{
    "microbit-dal": {
        "bluetooth": {
                    "enabled": 1,
                    "private_addressing": 0,
                    "advertising_timeout": 0,
                    "tx_power": 6,
                    "dfu_service": 1,
                    "device_info_service": 1,
                    "event_service": 1,
                    "eddystone_url": 1,
                    "eddystone_uid": 1,
                    "open": 0,
                    "pairing_mode": 1,
                    "whitelist": 1,
                    "security_level": "SECURITY_MODE_ENCRYPTION_NO_MITM",
                    "partial_flashing": 0
                },
        "gatt_table_size": "0x300"
    }
}
```

finneyj, 2018-09-08:
Cor. Stealth calibration. That's... unexpected.

finneyj, 2018-09-08:
Fixed in #dal-integration-5.

https://github.com/lancaster-university/microbit-dal/commit/ccd55f198d43c4026fc02ce23e64d0eb25c3e822
https://github.com/lancaster-university/microbit/commit/d54e26fd4d2c0812923f60874ea17df24432f636

closed 2018-09-08 by finneyj