lancaster-university/microbit-dal #385: Last LED at the end of compass calibration

microbit-carlos, 2018-09-14:
During the calibration procedure to fill the screen, one of the LEDs blinks to show the current orientation. If the LED happens to be OFF when fill-the-screen UX ends, it will stay off for a couple of seconds while the calibration calculations take place and until the smily face appears.

This could be confusing for users since it's not clear if the last pixel was filled or not.

finneyj, 2018-09-14:
@microbit-carlos 

Yes, I'd noticed that too. personally, I think this is a symptom of a slightly larger issue though, which is that the calibration algorithm now takes a few seconds longer to complete (because it's now based on iterative approximation), and there's no user feedback during this time that anything is happening.

I wonder if we should consider some simple "I'm working" type animation while the computation is taking place?

finneyj, 2018-09-14:
maybe something a bit like this:
https://makecode.microbit.org/_HvihLd026iPj

microbit-carlos, 2018-09-14:
An animation would probably, I think I would probably aim for whatever introduces the least overhead.

finneyj, 2018-09-14:
hie @microbit-carlos 

I'm afraid I'm not suite sure what you mean there? Were you thinking that an animation would add unnecessary resource overhead?

microbit-carlos, 2018-09-14:
Sorry, my last message there wasn't very eloquent. I meant that yes, an animation would help with that.  And that whatever animation we pick, I think the main factor for me would be to use whichever used less resources (not that a few images in flash take all that much space, but things like this add up).

finneyj, 2018-09-14:
Indeed. I think this warrants some user feedback, but we should keep it light.

you could always run a short competition out of the foundation to have kids design the best animation @microbit-carlos! Winner gets a microbit t-shirt and their animation used in the product!

jamesadevine, 2018-09-14:
> you could always run a short competition out of the foundation to have kids design the best animation @microbit-carlos! Winner gets a microbit t-shirt and their animation used in the product!

👍