lancaster-university/microbit-dal #388: Accelerometer z-axis value incorrect

helenlawes, 2018-09-23:
When reading the value of z-axis from the accelerometer my microbit is always returning -2048 regardless of how it is rotated. X and Y axis values do change. 

The only time the value changes is if I amend the accelerometer range then it reads:
-2048 for 1g and 2g
-4096 for 4g
-8192 for 8g

I've tried reading the value from the bluetooth service and I've also set it to display on the screen with the MakeCode blocks, using a forever block and show number acceleration (mg) z.

finneyj, 2018-09-25:
hi @helenlawes 

Thanks for reporting. Curious - i've not seen any other reports of this not working.

 I just tried this out on a few microbits here, and all looks ok...  I'm getting data on all three axes with this program, tested against the HEAD revision of the master branch here (v2.1.0) and a 1.3b version of the micro:bit (the most common one):

```
#include "MicroBit.h"

MicroBit uBit;

int main()
{
    // Initialise the micro:bit runtime.
    uBit.init();

    // Insert your code here!
    while(1)
    {
        uBit.serial.printf("[X:%d] [Y:%d] [Z:%d]\n", uBit.accelerometer.getX(), uBit.accelerometer.getY(), uBit.accelerometer.getZ());
        uBit.sleep(500);

    }
}
```

Can you see what version your micro:bit is? it is written vertically on the back of the device, at the bottom left corner edge, next to the P0 ring connector... if you let me know, I can test against the same version you have.

I'm wondering if you have a fault accelerator there. The accelerometer is a MEMS device (micro electro-mechanical system). I have heard of some people suffering for "sticky accelerometer syndrome", where one of the little bouncy balls-on-a-spring that is your accelerometer gets stuck at one extreme during manufacturing... You might be one of these (lucky?) few?

If you're feeling brave, you could try giving your micro:bit a good hearty whack on the desk to loosen it up a bit!

The Z axis is the one that runs perpendicular to the screen, so holding the device flat in your hand giving it a good smack on the front and back of the device has to be worth a go. :)

If that doesn't help, I think I'd be looking to exchange your micro:bit...

hope this helps!

helenlawes, 2018-09-25:
Hi @finneyj 

I've got the version 1.3b micro:bit. I gave the device a smack on the front and the back and this appears to have solved the issue :)

Many thanks

closed 2018-09-25 by helenlawes

finneyj, 2018-09-25:
Awesome!