lancaster-university/microbit-dal #392: Reset in pairing mode not available for radio programs

pelikhan, 2018-10-02:
This event is ignored when BLE is turned off.
```
MicroBitEvent evt(MICROBIT_ID_PARTIAL_FLASHING, MICROBIT_RESET );
```

pelikhan, 2018-10-02:
@jaustin @finneyj @microbit-sam this feature only works with a bluetooth program., can we get it always on?

jaustin, 2018-10-02:
Looks like at the moment this is only getting handled in the PartialFlashingService but could potentially be registered in uBit https://github.com/lancaster-university/microbit/blob/master/source/MicroBit.cpp#L201 

@microbit-sam what if we added a handler in uBit only if we're not also bringing up BLE? Looks like it relies on a StorageObject in BleManager at the moment hence the dependency but we do also have that in uBit. It might be a bit of duplication but we could possibly factor it out to a static method that takes a stroageobject in BLEManager?

microbit-sam, 2018-10-02:
ack, will implement tomorrow

pelikhan, 2018-10-02:
PR in MakeCode: https://github.com/Microsoft/pxt-microbit/pull/1329

pelikhan, 2018-10-04:
@microbit-sam ping

microbit-sam, 2018-10-05:
Sorry for the delay @pelikhan

I've moved it into uBit, however the existing event listener is listening for when other listeners are created. We can moved the reset into pairing mode code into this listener, but it means that the event required will be:

`MicroBitEvent(MICROBIT_ID_MESSAGE_BUS_LISTENER, MICROBIT_ID_RESET_INTO_PAIRING);`

Which is probably a bad/confusing idea.. and it doesn't look possible to specify multiple event sources

@finneyj is this a bad idea?

finneyj, 2018-10-05:
@microbit-sam you should be able to use a full wildcard and register the uBit listener as (MICROBIT_ID_ANY, MICROBIT_EVT_ANY),  but then all the events in the micro:bit's universe would get piped through that function. ..

This is a processing / RAM usage tradeoff I guess... and the micro:bit is relatively processor rich.

microbit-sam, 2018-10-05:
Cool, once the PRs submitted against microbit and microbit-dal are reviewed and merged you will be able to trigger a reset into pairing mode with a `MicroBitEvent` (The event source currently doesn't matter, but may be useful to know the source in the future)

Example:
```MicroBitEvent evt(MICROBIT_ID_PARTIAL_FLASHING, MICROBIT_ID_RESET_INTO_PAIRING );```

pelikhan, 2018-10-12:
I am also going to add back the "pairing mode" in our "open" setting in MakeCode. Otherwise, it's impossible to switch into this mode.

microbit-carlos, 2024-04-16:
To save resources we can implement this as a function call instead of an event.

In that case we need to check if MakeCode has got anything implemented already in this area.
This PR doesn't look like it was merged, but we need to double check:
https://github.com/microsoft/pxt-microbit/pull/1329/files#diff-c7001311f13221572261a0ad046afe85b7cdd4be0fe876c7ed1f4803bc403c71