lancaster-university/microbit-dal #396: Progress indication while (partial) flashing

pelikhan, 2018-10-12:
When the bit is in Pairing Mode and partial flashing is receiving packets, it would be great to show some kind of progress indicator (either blink system LED or use the screen). This would be VERY helpful.
We are in the pairing mode, so we can really do whatever we want with the screen.

@jaustin @finneyj @microbit-sam

jaustin, 2018-10-12:
I like this. As we discovered before animations are RAM-heavy so something
procedurally generated on the display would be ideal.

I'd rather show activity than progress, as that latter is notoriously hard
to do well, unless @microbit-sam you already know how large you expect the
program to be?

J

microbit-sam, 2018-10-12:
No, we currently don't know how large a file is, but could tell the m:b before setting off the partial flash.. would be nice to fill the screen up pixel by pixel until it's complete