lancaster-university/microbit-dal #404: Different file extension for binaries created with MakeCode?

smartyw, 2018-10-29:
Sometimes users import hex files into MakeCode. This works if the hex file was created with MakeCode in the first place but not if it wasn't (e.g. it was created from C/C++ and Yotta). When it doesn't work, this confuses users. MakeCode doesn't seem to give any feedback so they're left just thinking that nothing has happened.

I realise this is ostensibly a MakeCode issue but I wondered what this group think about perhaps having MakeCode create binaries with a different file extension like ".mkc" instead of ".hex" so it's possible to recognise those files which can be imported vs those which cannot.

Comments?

DavidWhaleMEF, 2018-10-29:
While this sounds like a good idea, it is sadly not recommended.

Many schools struggle to get new extensions approved via their third party IT providers, making this change would be quite disruptive for them.

It would be better for MakeCode to say 'this hex file was not written in MakeCode, please try these other editors [link]. A signpost at point of error is often better than making a change. Especially where schools and outsourced IT are involved!

DavidWhaleMEF, 2018-10-29:
Also @smartyw did you notice this is ticket number 404? Sorry, not found(!!!)

DavidWhaleMEF, 2018-10-29:
It's probably one we could log in the MakeCode github repo, the team there may just be able to tweak the (very short) error message that is displayed, to become a popup with some suggested alternative links perhaps?

pelikhan, 2018-10-29:
This is an issue to be raised in pxt-microbit - not a DAL issue.