lancaster-university/microbit-dal #406: Registering an EventHandler in a class

komodo108, 2018-11-02:
Below is an example of the problem, the MicroBit should print "Hi" whenever an action occurs, but is printing nothing.
```c++
#include "MicroBit.h"

class test {
    private:
        MicroBit* uBit;
    public:
        test(MicroBit* microbit) {
            uBit = microbit;
            uBit->messageBus.listen(MICROBIT_ID_GESTURE, MICROBIT_EVT_ANY, this, &test::event);
        }

        void event(MicroBitEvent e) {
            uBit->serial.send("Hi");
        }
};

MicroBit uBit;

int main() {
    uBit.init();
    test thetest(&uBit);
    while(true) {}
    release_fiber();
}
```

I am following the example method from [here](https://lancaster-university.github.io/microbit-docs/ubit/messageBus/).

pelikhan, 2018-11-02:
Remove ``while(true) {}``.

komodo108, 2018-11-02:
Sadly, that did not work.
Same behaviour as before.

finneyj, 2018-11-02:
Really? Did you keep the release_fiber() in there? That call deletes the main thread and releases any associated resources back to the system.

Removing the while(true) loop should be the way to go, or do a uBit.sleep() operation in there while loop (or anything else that blocks).  We're running a non-preemtpive scheduler to help keep things simpler for the kids...

finneyj, 2018-11-02:
OK - I can reproduce this. Something afoot. Looking into it...

komodo108, 2018-11-02:
@finneyj Sorry for the mistake! I was just following the tutorial from [here](https://lancaster-university.github.io/microbit-docs/), I removed changed the loop to:
`while(true) { uBit.sleep(10000); }` and have tried both with and without the call to `release_fiber()` and there is still no change when moving the MicroBit.

Just saw the second comment, and thank you :)

finneyj, 2018-11-02:
Something funky in the MessageBus event handling in this case I think, and/or the lazy instantiation of the gesture recognizer. If you try this (which is really just the same as your code with an addition C style handler, both event handlers trigger for me.)

```
#include "MicroBit.h"

class test {
    private:
        MicroBit* uBit;
    public:
        test(MicroBit* microbit) {
            uBit = microbit;
            uBit->messageBus.listen(MICROBIT_ID_GESTURE, MICROBIT_EVT_ANY, this, &test::event);
        }

        void event(MicroBitEvent e) {
            uBit->serial.printf("EVT2\n");
        }
};

MicroBit uBit;
test thetest(&uBit);

void allGestures(MicroBitEvent e) {
    uBit.serial.printf("EVT1\n");
}

int main() {
    uBit.init();
    uBit.messageBus.listen(MICROBIT_ID_GESTURE, MICROBIT_EVT_ANY, allGestures);

    while(1)
    {
        uBit.serial.send("Hi\n");
        uBit.sleep(2000);
    }
    release_fiber();
}
```

finneyj, 2018-11-02:
```
#include "MicroBit.h"

class test {
    private:
        MicroBit* uBit;
    public:
        test(MicroBit* microbit) {
            uBit = microbit;
            uBit->messageBus.listen(MICROBIT_ID_GESTURE, MICROBIT_EVT_ANY, this, &test::event);
        }

        void event(MicroBitEvent e) {
            uBit->serial.send("Hi");
        }
};

MicroBit uBit;

int main() {
    uBit.init();
    new test(&uBit);
    release_fiber();
}
```

komodo108, 2018-11-05:
Thanks so much!

closed 2018-11-05 by komodo108