lancaster-university/microbit-dal #417: v1.3 / v1.5 hardware differences

cpseager, 2019-01-10:
Hello, I tried to find this info but there doesn't seem to be much formal documentation about:

Am I correct in assuming:

1) predefined fixed hex files to do a certain job (e.g. talk to Scratch) as per here are for v1.3 boards only (although they don't say that)
https://scratch.mit.edu/microbit
  
2) they won't work with v1.5 boards, because of the accelerometer chip change, and just hang at 'draw a circle' type instructions

3) v1.5 are now on sale but customers have no idea whether they have 1.3 or v1.5?
 
Thanks!

whaleygeek, 2019-05-06:
1. Predefined hex files, if created in the older editors, will only have drivers for the separate accelerometer and magnetometer devices. If those hex files were created with the newer editors (from about October 2018), both old and new drivers are baked into the C++ DAL code, so both old and new boards will work seamlessly. A .hex file can have it's DAL drivers upgraded by dragging and dropping it into the appropriate editor then pressing DOWNLOAD.

2. See (1) above

3. There is a silk screen mark on the back of the board, component side, near the P0 pad. The original BBC distribution devices that were sent to schools didn't have a mark. micro:bits that were distributed after the Foundation was formed have a V1.3B mark, and the newer boards with the single combined accelerometer and compass have the V1.5 mark.

Hope this helps.

2.

whaleygeek, 2019-05-06:
Also these support articles are relevant:

https://support.microbit.org/support/solutions/articles/19000090748-updating-your-hex-file-for-the-motion-sensor-change

https://support.microbit.org/support/solutions/articles/19000087020-micro-bit-motion-sensor-change

Also see this page on the tech site:

https://tech.microbit.org/hardware/#motion-sensor

closed 2019-07-24 by cpseager