lancaster-university/microbit-dal #425: Magnetometer BLE service calibration states

ukBaz, 2019-02-23:
The calibration behaviour has been changed on the magnetometer... no auto-calibration when you instantiate the magnetometer service like previously. This results in no notifications or readings until calibration has been performed.
I can trigger a calibration by sending 0x01 to the new characteristic (e95db358-251d-470a-a062-fa1922dfa9a8).
I am writing Python on a Raspberry Pi to interact with the micro:bit. Not sure how to modify my code to handle this change. After issuing the 'calibrate' command, there does not seem a clean way to know when the calibration is finished. Also there doesn't seem to be a clean way to tell if this is the first time the device has been used and needs calibration. Anyone got any thoughts on this?

ukBaz, 2019-03-02:
Thanks @microbit-sam for updating the docs (as noted in https://github.com/lancaster-university/microbit-dal/issues/412#issuecomment-467529846) with some more information about calibration and values.
The docs now answer the question about knowing if the calibration has been successful or not.

It does not answer the question about how I detect if I need to calibrate the magnetometer or not.

The introduction of this calibration characteristic seems a backwards and unnecessary development.

When I use the magnetometer without Bluetooth I do not have manage if the device needs calibration. If I use the heading value the device promotes for calibration if it needs it. Previously that is what the Bluetooth magnetometer service also did. When I started the service it would prompt the user to calibrate the magnetometer if it required it.
Why has this changed to a manual step?  It seems to complicate the code I need to write for no gain and is inconsistent with the non-Bluetooth mode usage.

I have tried to have a discussion about this on Slack and now here. 
To ask a direct question:
Is there a possibility that we can revert the Bluetooth magnetometer service so it manages if it needs calibration or not?
@jaustin is this something you can answer?

microbit-sam, 2019-03-04:
Hi Barry, sorry I'm not very active on Slack so will have missed any discussion on there

This commit prevents the compass from automatically triggering calibration, I think (may be misremembering) this was added so for cases where a device has a malfunctioning mag/acc would not be able to get past the calibration screen, although if that is the case perhaps a timeout / escape key would be better.. @finneyj is that right?
https://github.com/lancaster-university/microbit/commit/d54e26fd4d2c0812923f60874ea17df24432f636#diff-6047755bfccc64c91f0d390dd2f1061e

Calling `compass.heading()` should trigger a calibration, or `compass.calibrate()` immediately after creating the MagnetometerService could be a possible workaround for now