lancaster-university/microbit-dal #432: Two different serial numbers?

smartyw, 2019-03-22:
On my device, microbit_serial_number() returns 2117738716 but the Bluetooth device information service, due to the function used to set it, contains 841232314.

The Bluetooth Device Information Service Serial Number characteristic is set when initialising with

```
bleManager.init(getName(), getSerial(), messageBus, true);
```
getSerial is
```
inline ManagedString MicroBit::getSerial()
{
    // We take to 16 bit numbers here, as we want the full range of ID bits, but don't want negative numbers...
    int n1 = microbit_serial_number() & 0xffff;
    int n2 = (microbit_serial_number() >> 16) & 0xffff;

    // Simply concat the two numbers.
    ManagedString s1(n1);
    ManagedString s2(n2);

    return s1 + s2;
}
```

Why is the Bluetooth serial number populated in this way and not just with microbit_serial_number()?

martinwork, 2019-04-11:
Would anything break if we changed the dev info number to  2117738716?

bluetooth-mdw, 2019-04-11:
I can't see any reason why it would.... I personally don't ever use the serial number other than to display it. I guess if someone was assuming it would never change and using it to identify specific devices, they'd have a problem.... but that does seem unlikely. Probably bt_addr is most typically used, especially as we don't use resolvable private addresses. Educated guesswork at best, though.

bluetooth-mdw, 2019-04-11:
That said, I do wonder what the background to this is.... maybe there was a good reason for doing it this way? Maybe @jamesadevine remembers?

jamesadevine, 2019-04-11:
Absolutely no idea Martin, apologies.

bluetooth-mdw, 2019-04-11:
@jamesadevine You just can't get the staff ;-)

Sorry, I thought I'd seen your name in some of that code. Maybe not.

jamesadevine, 2019-04-11:
Git blame me baby! 

I used to write occasionally useful commit messages so have a look (I’m away atm so have no ability to check myself!)

bluetooth-mdw, 2019-04-11:
Git credit you, more like. I imagine there's a good reason for this.... I just have no idea what it is!

martinwork, 2019-04-11:
I think it appeared here https://github.com/lancaster-university/microbit-dal/commit/db3eccf6ff4b9515602506a7445edbe72b625dda, long, long ago, courtesy of @finneyj.