lancaster-university/microbit-dal #433: Official off-board access to LED matrix

whaleygeek, 2019-03-22:
We need an official, DAL-supported method of accessing the LED matrix from the micro:bit, so we can build Mega:Bit's.

There is a fork of this repo (@jaustin can you link to it here please as I cannot find it) where some students at Imperial College did an I2C interface to an external LED display. This fork needs to be merged into mainline, and pushed into both MicroPython and MakeCode releases.

We also need an ETA of when this work will be completed, so that third parties can design and build Mega:Bits.

In the meantime, soldering on to LEDs is the best we have... documented here:

http://www.pocketmoneytronics.co.uk/?page_id=398

microbit-mark, 2019-03-22:
Hey @whaleygeek do you mean this one https://github.com/LukeB101/Mega-Bit/tree/master/Final%20Product%20Design/Software/modified-DAL ?

whaleygeek, 2019-03-24:
Yes @microbit-mark that's the one.

Is there an ETA on when this work will surface into MicroPython and MakeCode? I saw @jaustin on slack the other day saying it was being looked in to at the moment:

````
We’re hoping to try and make a more ‘official’ way of doing this in the DAL
```

The official way to do this would be to pull in the changes from that repo.

Has @LukeB101 sent a PR yet that @finneyj can merge?