lancaster-university/microbit-dal #435: Problem with ADC on Micro:Bit card

laboiteaphysique, 2019-03-28:
Hi !
I've send this message on the Micro:Bit Support page and the technical support engineer asks me to contact the DAL developpers on Github 
"I'm using the the analog input on the device,  and I was surprised by the difference between the analog value returned by the microbit and the "real" value on the P0 pin. (see attach file).
I've just found the reason :
the reference voltage for the embedded ADC comes from the VDD ... and VDD is not équal to 3.3V (when powered by USB) but less than 3,3V because of the Schottky diode placed on the power rail. It means that the VDD is around 3.2V (could change with the current consumption) ... and this is the value to keep for the calculation (and not 3.3V).
I've tried powering the microbit card with a trimmable external power  : it confirms this fact : for the same voltage applied on the P0 input, the ADC code conversion increase when I decrease VDD !
When working with USB supply thats not so much a problem if the notice doesn't give 3.3V as the maximum value, but when powered with battery it could be a serious drawback if the voltage applied  on the P0 pin (for example) is not a fraction of the VDD microbit supply, but comes from an another supply : the ADC code conversion would depend on the voltage given by the batteries (and would change all along the batteries life) !
I wonder why such a choice for the ADC reference voltage ... Why not choosing the internal Vref bandgap 1,2V ?

(For information : the attached file
[Problem_ADC.pdf](https://github.com/lancaster-university/microbit-dal/files/3019708/Problem_ADC.pdf)  is an excerpt from one of the free books I'm writting in french 
 "Discovering Python while programming a microcontroller" and the second about "Extend the functionalities of a microcontroller" is started . We can find them on my site : https://laboiteaphysique.fr/site2/index.php )
Best regards
Vincent

whaleygeek, 2019-05-06:
I don't recall the reason for this, @jaustin or @jamesadevine may remember.

But I suspect it was so that the common use-case of a child attaching a potential divider circuit (3V, 0V, P0 for example) always gave a value between 0 and 1023 (ADC_MAX), and that would be the case for both powered from USB, and powered from AAA batteries with a range of state of charge. 

We see a wide range of voltages from AAA batteries, and a slight increase in voltage when powered from USB, so any 'fixed calibration' (threshold value) used in the child's code could vary significantly when using the (most common) potential divider setup. This would make their code perform differently in different battery conditions and also when powered from USB.

The use-case of reading an external voltage on the micro:bit is less common with children (and over-voltage-ing is a real risk if powered externally).

You can change the reference configuration, here is an example that @microbit-carlos wrote for us in MicroPython

https://gist.github.com/microbit-carlos/baebc389bccc4366abf3701acc7fc61c

laboiteaphysique, 2019-05-11:
OK, thanks for this answer !
I'm interested in the Micro:Bit card because of its price and the possibilities. Unfortunately it sometimes appears like a too much "closed" device dedicated to a beginner's usage. Intermediate and advanced users have to search a lot or to experiment to get good information. 
Best regards
Vincent

whaleygeek, 2019-05-11:
There is a lot of tech info about the micro:bit at https://tech.microbit.org

Also you can sign up for a developer.mbed.org account and code it in C++.

All of the low layer C++ and C code is open source here in this repo and in the mbed repos, and Daplink (which is used for the interface chip) is also fully open source, as well as the makecode editor and both our web and offline python IDE's - all the links to the various repos are on the tech site, along with detailed data-sheets for all chips used on the board, and also links to the hardware schematics (also open source) here: https://github.com/bbcmicrobit/hardware

Hope this helps!

jaustin, 2019-05-13:
@whaleygeek has given you some brilliant pointers (Thanks as always, David!). The other sites I find myself using a lot when using C/C++  on micro:bit are
* https://lancaster-university.github.io/microbit-docs/
* https://developer.mbed.org/platforms/microbit/ and associated pages

For example, here's an mbed example that configures the ADC to be absolute: 
https://os.mbed.com/teams/microbit/code/microbit-battery-test/
(specifically https://os.mbed.com/teams/microbit/code/microbit-battery-test/file/6c9081f485dd/main.cpp/ )

For the task you're interested in we also have #356. @laboiteaphysique would you be happy for me to close this issue in favour of 356, which I think is asking for the same thing?

laboiteaphysique, 2019-05-13:
Hi !
Thanks for all of those informations !
I close this issue

closed 2019-05-13 by laboiteaphysique