lancaster-university/microbit-dal #447: Missing public method accelerometer.whoAmI()

bobwilmes, 2019-06-26:
The file microbit-dal/source/drivers/MicroBitAccelerometer.cpp appears to be missing a declaration for the public method accelerometer.whoAmI() as described in the documentation here:
https://lancaster-university.github.io/microbit-docs/ubit/accelerometer/#whoami

martinwork, 2019-06-26:
I think this is the case. In fact the whoAmI function is not implemented yet. PR #373 was to implement whatAmI.

From https://lancaster-university.github.io/microbit-docs/ubit/accelerometer/
"There are two variants of the micro:bit , one uses the MMA8653 and the other a uses the LSM303 combined accelerometer and magnetometer."

If you need to determine which accelerometer is in use now, as a temporary workaround, you could try using the isDetected functions. These functions make a call to I2C to see what hardware is present, whereas whoAmI/whatAmI would return the previously detected state.

For example,
```
#include "LSM303Accelerometer.h"
int isLSM303 = LSM303Accelerometer::isDetected(uBit.i2c);
```

https://github.com/lancaster-university/microbit-dal/blob/9164922a519aa213a565cbf2abaa2cf4debf4977/inc/drivers/MMA8653.h#L124

https://github.com/lancaster-university/microbit-dal/blob/9164922a519aa213a565cbf2abaa2cf4debf4977/inc/drivers/LSM303Accelerometer.h#L148

https://github.com/lancaster-university/microbit-dal/blob/9164922a519aa213a565cbf2abaa2cf4debf4977/inc/drivers/FXOS8700.h#L225

bobwilmes, 2019-06-26:
Thank you Martin - I tried the work around and it works.  Slight nit - on the public class names for the accelerometer - the MMA8653 and the FXOS8700 use class name's based on the device as in
<device-name>::isDetected. 

The LSM303 uses LSM303Accelerometer::isDetected.

martinwork, 2019-06-27:
Not sure I understand, but yes, we should only need to use functions in MicroBitAccelerometer (ie. uBit.accelerometer). Accessing the device specific class is the only workaround I can see, until whoAmI gets added.