lancaster-university/microbit-dal #456: Question - High Drive GPIO pins

mastertechisme, 2019-12-03:
Does the DAL set any of the GPIO pins to high drive, 5ma, instead of .5 ma?  I see examples of micro:bit driving an LED directly of pin 0 with a max of 5 ma.  So somewhere I assume this is being set because makecode does not have a function to set high drive.

whaleygeek, 2019-12-04:
When I queried Nordic about this (around November 2016) the answer was 'no' (their engineers even ran the code in a debugger and checked the values in the pin drive registers, and confirmed 5mA high drive was disabled).

You will have to check with @jamesadevine as to whether any changes have been made to the DAL since then that affect this configuration.

There is an article on the support site here that shows how to write to chip registers from user code, along with a link to the device register map, so you could change the settings yourself and see if it allows you to achieve your current goal, whatever that may be.

https://support.microbit.org/support/solutions/articles/19000070728-how-to-read-the-device-serial-number
 
Note also some analysis that @edbye did a while back, shows that GPIO pins will provide more than the stated/configured current limit. This was confirmed by a Nordic engineer at the time, who said that GPIO pins will continue to provide more current, but that the voltage will de-rate (roughly linearly) and therefore the pin voltage specs cannot be guaranteed or maintained at higher currents (and thus a logic '1' level will not be guaranteed any more). Also the performance of analogs and the radio block would suffer as a result. 

@edbye discovered by systematic testing, that the GPIO pin will continue to de-rate it's voltage until a point where the pin becomes temporarily non functional, eventually the GPIO block shuts down completely, both are recovered by power cycling the device.

Hope this helps.

jamesadevine, 2019-12-04:
@whaleygeek Thanks David! I investigated all the various layers last night out of curiosity and couldn't see any place where high drive mode is set. I also looked at the nrf51822 product specification and noted that the maximum current draw for a pin is not specified, so I'm glad your findings confirm this.

@mastertechisme do you require access to high drive?

martinwork, 2019-12-04:
Sorry if this is a red herring. This is confusing if high drive mode is disabled: 
https://support.microbit.org/support/solutions/articles/19000101863-connecting-an-led-to-the-micro-bit
"Each pin of the micro:bit also has a current supply limit of 5mA and a maximum of 3 pins can be configured in this high drive mode at one time.".

whaleygeek, 2019-12-04:
Hi Martin,

It's a good point you make. However although it might be a tad misleading, it is still correct - it does not say whether the pins are pre-configured in that mode or not.

But we should open a github ticket on the tech site to clarify, I think...

https://github.com/microbit-foundation/dev-docs/issues

edbye, 2019-12-05:
Not sure I clearly understand what the question(s) is.
About 3 year ago I did some characterisation of normal (0.5mA) drive on GPIO outs., with results as mentioned by @whaleygeek 
The GPIO outs are fullly protected against overload inc short circuit.
The Nordic info in simple terms means that if the values are exceed then the device specification will not be met and unreliable results can be expected.
How to successfully drive LED's from the NRF51822 as presented in much of the write-ups could benefit from some clarity based on how LED's work and also what is to be acheived by the circuit.

mastertechisme, 2019-12-05:
Yes, let me clarify.
I am trying to get specifics on GPIO maximum current.  For example, my students may want to power LEDs or relays, which have a higher drive current from GPIO.  Relay needs 3ma to drive opto coupler to enable relay, LED's if powered from GPIO could need up to 20 ma, but controlled by resistor.

On an Arduino, you get an overall current budget for all GPIO combined, 100ma lets say, plus a maximum of 20ma per GPIO.  At least here I can tell what is possible with some math.

On the microbit, I see several mentions that GPIO can source 5ma, but only 3 pins, if enabled through software.  
https://support.microbit.org/support/solutions/articles/19000101863-connecting-an-led-to-the-micro-bit
I did not see any command in makecode to set high drive mode on any pin.

So I am looking for specifics on current from GPIO pins.  If you have a link to your write up that would be great.  I also see from that article on Microbit site they are pulling 5 ma from pin 0.  I was also wondering are pins 0,1,2 set to 5ma by default?

This is rationale behind the question, hope this helps.
Thanks for all the replies.

edbye, 2019-12-06:
The intended purpose for ‘high drive’ is to maintain correct logic levels when interfacing with other logic IC’s, not for driving things like LED’s.

Changing from standard GPIO drive to high will not change the published limitations on current capabilities. It will not give an overall increase or per pin capability.

In standard drive you can comfortably drive up to 5mA, as evidenced by the Microbit LED matrix, 220R resistor to Red LED.

If high current GPIO output is required, the correct way is to use a hardware interface.

When it comes to LED’s, why do you need 20mA? Most, if not all LED’s will give a very reasonable level of illumination with a current of 2mA or less.

Hope that clarifiies and helps

martinwork, 2019-12-06:
If the capability to drive up to 5mA is not related to high drive, does that mean the mention of high drive and a 3 pin limit is not relevant in the article about connecting LEDs?

whaleygeek, 2019-12-06:
Not really (i.e. it is still relevant). 5mA high drive, if enabled, will guarantee up to 5mA for each of the 3 configured pins *while* maintaining guaranteed voltage specifications such that VOH and VOL logic levels are within spec for logic circuits also. In the default case of std drive, the VOH in particular will linearly collapse as you draw more current, so you cant guarantee that attached logic circuits will also work in those conditions.

edbye, 2019-12-06:
having done some more research on this, I stand corrected. In high drive mode, you can get substantially more than 15mA output current from a GPIO https://devzone.nordicsemi.com/f/nordic-q-a/19897/nrf51822-gpio-direct-source-with-high-drive-setting

As @whaleygeek says, it is the chip specifications that can not be met above 5mA (high drive), for standard drive it is less.

The question is does this matter? This can only be answered by considering the circuitry requirements of what is being driven i.e. the appropriate datasheet must be consulted for respective VIH and VIL over the range of the power rail voltage.

The degredation of VOH only applies to the GPIO under consideration, each of the GPIO's has its own monitoring circuit.