lancaster-university/microbit-dal #460: MicroBitSerial.cpp   if(newHead != rxBuffTail)  This may result in loss of received data.

zfm076, 2020-03-11:
```
void MicroBitSerial::dataReceived()
{
    //get the received character
    //Note: always read from the serial to clear the RX interrupt
    char c = getc();

    if(!(status & MICROBIT_SERIAL_RX_BUFF_INIT))
        return;

    int delimeterOffset = 0;
    int delimLength = this->delimeters.length();

    //iterate through our delimeters (if any) to see if there is a match
    while(delimeterOffset < delimLength)
    {
        //fire an event if there is to block any waiting fibers
        if(this->delimeters.charAt(delimeterOffset) == c)
            MicroBitEvent(MICROBIT_ID_SERIAL, MICROBIT_SERIAL_EVT_DELIM_MATCH);

        delimeterOffset++;
    }

    uint16_t newHead = (rxBuffHead + 1) % rxBuffSize;

    //look ahead to our newHead value to see if we are about to collide with the tail
    if(newHead != rxBuffTail)
    {
        //if we are not, store the character, and update our actual head.
        this->rxBuff[rxBuffHead] = c;
        rxBuffHead = newHead;

        //if we have any fibers waiting for a specific number of characters, unblock them
        if(rxBuffHeadMatch >= 0)
            if(rxBuffHead == rxBuffHeadMatch)
            {
                rxBuffHeadMatch = -1;
                MicroBitEvent(MICROBIT_ID_SERIAL, MICROBIT_SERIAL_EVT_HEAD_MATCH);
            }
    }
    else
        //otherwise, our buffer is full, send an event to the user...
        MicroBitEvent(MICROBIT_ID_SERIAL, MICROBIT_SERIAL_EVT_RX_FULL);
}
```
I found a problem during the serial communication test: My computer sends data continuously through the serial port ( baud(57600) ) and no delay. and then my microbit just do one thing:The data is read and sent to the computer: 
```
        while(uBit.serial.isReadable())
        {
            data = uBit.serial.read(ASYNC);
            uBit.serial.send( &data ,1, SYNC_SLEEP);
        }
```
Then,I found that a lot of the data sent to the computer had been lost. I see the source inside this sentence : if(newHead != rxBuffTail). that  will cause my data to be lost. After I mask this code (if(newHead != rxBuffTail)), the data will still be lost, but much less.Is there any way to ensure the integrity of my data?Even if it overwrites the original data.

martinwork, 2020-03-11:
Related? https://github.com/lancaster-university/microbit-dal/issues/441

zfm076, 2020-03-12:
> Related? #441

This problem occurs when the queue is full.My computer has been sending this sentence(0xFF 0x01 0x02 0x03 0x04 0x05 0x06 0xAA) very quickly.Then microbit prints the received data as follows:
`FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 01 02 03 04 05 06 AA FF 02 FF 01 FF 01 02 FF 01 02 FF 01 FF 01 02 FF 01 03 FF 01 FF 01 FF 01 02 05 FF 03 FF 01 FF 01 02 FF 01 FF 01 02 06 FF FF 01 02 FF 01 02 FF 01 FF 01 02 FF 01 02 FF 01 05 FF 01 AA FF 01 FF 01 03 FF 02 FF 01 FF 01 02 FF 01 FF 01 02 FF 01 02 FF 01 FF 01 02 FF 01 FF 01 02 05 FF 01 02 FF 01 FF 01 FF 01 FF 01 02 FF 01 02 03 FF 01 FF 01 02 FF FF 01 02 05 FF 01 FF 01 FF 01 02 FF 01 02 FF 01 05 FF 01 FF 01 FF 01 02 FF 01 AA FF 01 02 FF 01 FF 01 AA FF 01 FF 01 02 FF 01 06 FF 01 FF 01 AA FF 01 05 FF 01 FF 01 03 FF 01 FF 01 06 FF 01 FF 01 02 FF 01 05 FF 01 FF 01 02 FF 01 FF 01 FF 01 02 03 FF 01 FF 01 04 FF 01 FF 01 06 FF 01 FF 01 02 06 FF 04 FF 01 FF 01 02 FF 01 02 FF 01 FF 01 FF 01 02 05 FF 01 FF 01 05 FF 01 FF 01 05 FF 01 FF 01 AA FF 01 FF 01 02 AA FF 01 FF 01 FF 01 02 FF 01 AA FF 01 02 FF 04 FF 01 FF 01 FF 01 02 AA FF 01 02 FF AA FF 01 FF 01 FF 01 02 06 FF 01 FF 01 05 FF 01 FF 01 02 FF 01 FF 01 02 FF 01 02 FF 01 02 FF 02 FF 01 02 FF 02 FF 01 AA FF 02 FF 01 02 FF 01 02 FF 01 FF 01 05 FF 01 FF 01 02 FF 01 06 FF 01 FF 01 02 FF 01 FF 01 02 FF 01 FF 01 02 FF 01 AA FF 01 FF 01 AA FF 01 02 FF 01 FF 01 03 FF 01 FF 01 04 FF 01 FF 01 02 06 FF 01 FF 01 02 FF 05 FF  `
You can see that the data is normal when the queue is not full.When the queue is full, the data is cut off.I know the data will be lost, but it shouldn't be cut off.

martinwork, 2020-03-12:
When the buffer is full, incoming bytes are dropped. When read() removes one byte, the next incoming byte will be added.

By commenting out the buffer full check, incoming bytes are overwriting old data. Data is still being lost. Reading will potentially get recent bytes followed by older ones.

Try sending frames with a counter, then see how the data looks:
0xFF 0x01 0x02 0x03 0x04 0x05 0x06 0xAA,
0xFF 0x02 0x02 0x03 0x04 0x05 0x06 0xAA,
0xFF 0x03 0x02 0x03 0x04 0x05 0x06 0xAA,
etc.

Serial works one byte at a time. Maybe set the buffer size to be a multiple of 8, check the size of the data available and read in multiples of 8.

zfm076, 2020-03-13:
> When the buffer is full, incoming bytes are dropped. When read() removes one byte, the next incoming byte will be added.
>
> By commenting out the buffer full check, incoming bytes are overwriting old [...]

My microbit code:
```
#include "MicroBit.h"
MicroBit uBit;
int main()
{
    uBit.init();
    uBit.serial.baud(57600);
    uBit.serial.setRxBufferSize(200);
    uBit.serial.setTxBufferSize(200);
   
    while(1)
   {
        while(uBit.serial.isReadable())
        {
            uBit.serial.read( &data ,1, ASYNC);
            uBit.serial.send( &data ,1, ASYNC);
            uBit.sleep(0.05);//I deliberately increase the delay to let the serial port receive queue fill up, 
                                         in order to see what happens
        }
    }
    release_fiber();
}
```
I take your advice,sending frames with a counter,My computer sent 255 frames of data,Then microbit prints the received data as follows:
`AA 01 02 03 04 05 06 FF AA 02 02 03 04 05 06 FF AA 03 02 03 04 05 06 FF AA 04 02 03 04 05 06 FF AA 05 02 03 04 05 06 FF AA 06 02 03 04 05 06 FF AA 07 02 03 04 05 06 FF AA 08 02 03 04 05 06 FF AA 09 02 03 04 05 06 FF AA 0A 02 03 04 05 06 FF AA 0B 02 03 04 05 06 FF AA 0C 02 03 04 05 06 FF AA 0D 02 03 04 05 06 FF AA 0E 02 03 04 05 06 FF AA 0F 02 03 04 05 06 FF AA 10 02 03 04 05 06 FF AA 11 02 03 04 05 06 FF AA 12 02 03 04 05 06 FF AA 13 02 03 04 05 06 FF AA 14 02 03 04 05 06 FF AA 15 02 03 04 05 06 FF AA 16 02 03 04 05 06 FF AA 17 02 03 04 05 06 FF AA 18 02 03 04 05 06 FF AA 19 02 03 04 05 06 FF AA 1A 02 03 04 05 06 FF AA 1B 02 03 04 05 06 FF AA 1C 02 03 04 05 06 FF AA 1D 02 03 04 05 06 FF AA 1E 02 03 04 05 06 FF AA 1F 02 03 04 05 06 FF AA 20 02 03 04 05 06 FF AA 21 02 03 04 05 06 FF AA 22 02 03 04 05 06 FF AA 23 02 03 04 05 06 FF AA 24 02 03 04 05 06 FF AA 25 02 03 04 05 06 FF AA 26 02 03 04 05 06 FF AA 27 02 03 04 05 06 FF AA 28 02 03 04 05 06 FF AA 29 02 03 04 05 06 FF AA 2A 02 03 04 05 06 FF AA 2B 02 03 04 05 06 FF AA 2C 02 03 04 05 06 FF AA 2D 02 03 04 05 06 FF AA 2E 02 03 04 05 06 FF AA 2F 02 03 04 05 06 FF AA 30 02 03 04 05 06 FF AA 31 02 03 04 05 06 FF AA 32 02 03 04 05 06 FF AA 33 02 03 04 05 06 FF AA 34 02 03 04 05 06 FF AA 35 02 03 04 05 06 FF AA 36 02 03 04 05 06 FF AA 37 02 03 04 05 06 FF AA 38 02 03 04 05 06 FF AA 39 02 03 04 05 06 FF AA 3A 02 03 04 05 06 FF AA 3B 02 03 04 05 06 FF AA 3C 02 03 04 05 06 FF AA 3D 02 03 04 05 06 FF AA 3E 02 03 04 05 06 FF AA 3F 02 03 04 05 06 FF AA 40 02 03 04 05 06 FF AA 41 02 03 04 05 06 FF AA 42 02 03 04 05 AA 43 02 03 04 AA 44 02 03 04 AA 45 02 03 04 AA 46 02 03 04 AA 47 02 03 04 AA 48 02 03 04 AA 49 02 03 04 AA 4A 02 03 04 AA 4B 02 03 04 AA 4C 02 03 04 AA 4D 02 03 04 AA 4E 02 03 04 AA 4F 02 03 04 AA 50 02 03 04 AA 51 02 03 04 AA 52 02 03 04 AA 53 02 03 04 AA 54 02 03 04 AA 55 02 03 04 AA 56 02 03 04 AA 57 02 03 04 AA 58 02 03 04 AA 59 02 03 04 AA 5A 02 03 04 AA 5B 02 03 04 AA 5C 02 03 04 AA 5D 02 03 04 AA 5E 02 03 04 AA 5F 02 03 04 AA 60 02 03 04 AA 61 02 03 04 AA 62 02 03 04 AA 63 02 03 04 AA 64 02 03 04 AA 65 02 03 04 AA 66 02 03 04 AA 67 02 03 04 AA 68 02 03 04 AA 69 02 03 04 AA 6A 02 03 04 AA 6B 02 03 04 AA 6C 02 03 04 AA 6D 02 03 04 AA 6E 02 03 04 AA 6F 02 03 04 AA 70 02 03 04 AA 71 02 03 04 AA 72 02 03 04 AA 73 02 03 04 AA 74 02 03 04 AA 75 02 03 04 AA 76 02 03 04 AA 77 02 03 04 AA 78 02 03 04 AA 79 02 03 04 AA 7A 02 03 04 AA 7B 02 03 04 AA 7C 02 03 04 AA 7D 02 03 04 AA 7E 02 03 04 AA 7F 02 03 04 AA 80 02 03 04 AA 81 02 03 04 AA 82 02 03 04 AA 83 02 03 04 AA 84 02 03 04 AA 85 02 03 04 AA 86 02 03 04 AA 87 02 03 04 AA 88 02 03 04 AA 89 02 03 04 AA 8A 02 03 04 AA 8B 02 03 04 AA 8C 02 03 04 AA 8D 02 03 04 AA 8E 02 03 04 AA 8F 02 03 04 AA 90 02 03 04 AA 91 02 03 04 AA 92 02 03 04 AA 93 02 03 04 AA 94 02 03 04 AA 95 02 03 04 AA 96 02 03 04 AA 97 02 03 04 AA 98 02 03 04 AA 99 02 03 04 AA 9A 02 03 04 AA 9B 02 03 04 AA 9C 02 03 04 AA 9D 02 03 04 AA 9E 02 03 04 AA 9F 02 03 04 AA A0 02 03 04 AA A1 02 03 04 AA A2 02 03 04 AA A3 02 03 04 AA A4 02 03 04 AA A5 02 03 04 AA A6 02 03 04 AA A7 02 03 04 AA A8 02 03 04 AA A9 02 03 04 AA AA 02 03 04 AA AB 02 03 04 AA AC 02 03 04 AA AD 02 03 04 AA AE 02 03 04 AA AF 02 03 04 AA B0 02 03 04 AA B1 02 03 04 AA B2 02 03 04 AA B3 02 03 04 AA B4 02 03 04 AA B5 02 03 04 AA B6 02 03 04 AA B7 02 03 04 AA B8 02 03 04 AA B9 02 03 04 AA BA 02 03 04 AA BB 02 03 04 AA BC 02 03 04 AA BD 02 03 04 AA BE 02 03 04 AA BF 02 03 04 AA C0 02 03 04 AA C1 02 03 04 AA C2 02 03 04 AA C3 02 03 04 AA C4 02 03 04 AA C5 02 03 04 AA C6 02 03 04 AA C7 02 03 04 AA C8 02 03 04 AA C9 02 03 04 AA CA 02 03 04 AA CB 02 03 04 AA CC 02 03 04 AA CD 02 03 04 AA CE 02 03 04 AA CF 02 03 04 AA D0 02 03 04 AA D1 02 03 04 AA D2 02 03 04 AA D3 02 03 04 AA D4 02 03 04 AA D5 02 03 04 AA D6 02 03 04 AA D7 02 03 04 AA D8 02 03 04 AA D9 02 03 04 AA DA 02 03 04 AA DB 02 03 04 AA DC 02 03 04 AA DD 02 03 04 AA DE 02 03 04 AA DF 02 03 04 AA E0 02 03 04 AA E1 02 03 04 AA E2 02 03 04 AA E3 02 03 04 AA E4 02 03 04 AA E5 02 03 04 AA E6 02 03 04 AA E7 02 03 04 AA E8 02 03 04 AA E9 02 03 04 AA EA 02 03 04 AA EB 02 03 04 AA EC 02 03 04 AA ED 02 03 04 AA EE 02 03 04 AA EF 02 03 04 AA F0 02 03 04 AA F1 02 03 04 AA F2 02 03 04 AA F3 02 03 04 AA F4 02 03 04 AA F5 02 03 04 AA F6 02 03 04 AA F7 02 03 04 AA F8 02 03 04 AA F9 02 03 04 AA FA 02 03 04 AA FB 02 03 04 AA FC 02 03 04 AA FD 02 03 04 AA FE 02 03 04 AA FF 02 03 04 `

Then By commenting out the buffer full check,The microbit prints the received data as follows:
[... cut ...] 05 06 FF AA F5 02 03 04 05 06 FF AA F6 02 03 04 05 06 FF AA F7 02 03 04 05 06 FF AA F8 02 03 04 05 06 FF AA F9 02 03 04 05 06 FF AA FA 02 03 04 05 06 FF AA FB 02 03 04 05 06 FF AA FC 02 03 04 05 06 FF AA FD 02 03 04 05 06 FF AA FE 02 03 04 05 06 FF AA FF 02 03 04 05 06 FF `

By comparison, I prefer choice commenting out the buffer full check.Because even if the data is lost, the data frame is more complete.But I don't know what happens to the other code if I commenting out the buffer full check.

zfm076, 2020-03-16:
> When the buffer is full, incoming bytes are dropped. When read() removes one byte, the next incoming byte will be added.
>
> By commenting out the buffer full check, incoming bytes are overwriting old [...]

Hi, martinwork.  what happens to the other code if I commenting out the buffer full check？

martinwork, 2020-03-16:
I'm sorry, I don't understand what you're asking. Did you reading 8 bytes at a time?

zfm076, 2020-03-17:
> I'm sorry, I don't understand what you're asking. Did you reading 8 bytes at a time?

 I reading 1 bytes at a time (uBit.serial.read( &data ,1, ASYNC);). Because it was the computer communicating with microbit, When my microbit handle serial port data, I judge the header bytes(0xAA) first. then The data is read until the end of the byte(0xFF) is read , Read through one frame and then process one frame of data.  If a byte is lost in transit，I might not be able to read the correct data frame by reading 8 bytes. So I can't read 8 bytes at a time.