lancaster-university/microbit-dal #461: serial/fibers problem

mmoskal, 2020-04-01:
From https://github.com/microsoft/pxt-microbit/issues/2709

The following program blinks nicely until A is pressed. Then the release_fiber() seems to exit.

Joe, this is blocking on MakeCode side.

```cpp
void onButton(MicroBitEvent e) {
    uBit.serial.send((uint8_t *)" ", 1);
}

void blink() {
    for (;;) {
        uBit.display.image.setPixelValue(0, 0, 255);
        fiber_sleep(100);
        uBit.display.image.setPixelValue(0, 0, 0);
        fiber_sleep(100);
    }
}

int main() {
    uBit.init();
    uBit.messageBus.listen(MICROBIT_ID_BUTTON_A, MICROBIT_BUTTON_EVT_CLICK, onButton);
    create_fiber(blink);
    release_fiber();
    return 0;
}
```

finneyj, 2020-04-01:
Thanks @mmoskal. I'll check it out today.

finneyj, 2020-04-01:
Fixed in https://github.com/lancaster-university/microbit-dal/commit/f96a39036664495c7f934d1688c744f7d51b025b

zfm076, 2020-04-03:
> Fixed in [f96a390](https://github.com/lancaster-university/microbit-dal/commit/f96a39036664495c7f934d1688c744f7d51b025b)

Hi, I found a problem on the radio.Could you take a look at it for me?[https://github.com/lancaster-university/microbit-dal/issues/462](url)

commit f96a3903 (2020-04-01): Use schedule() not fiber_sleep(0) in MicroBitSerial.cpp (Fix #461)

  - MicroBitSerial use wake_on_event for condition synchronisation.
  - By contract, a fiber doing this should subsequently call schedule()
    and no other blocking operation