lancaster-university/microbit-dal #463: printChar does not block

pelikhan, 2020-04-17:
This code in MakeCode leads to a blank screen because X is erased immediately by clearScreen
```
input.onButtonPressed(Button.A, function () {
    basic.showString("X")
    basic.clearScreen()
})
```
The showString function in PXT calls into printChar.

https://github.com/microsoft/pxt-microbit/blame/50ec105c4faeaca8bf1ac8636b5b5021afd937ba/libs/core/basic.cpp#L37

Could this be related to the new fibers?

pelikhan, 2020-04-20:
@finneyj @jamesadevine

finneyj, 2020-04-21:
@pelikhan 

Thanks - Fixed in v2.2.0-rc4

I went through the while print() family of methods and tested/unified behaviour, so hopefully we won't see any more regressions in MicroBitDisplay.

pelikhan, 2020-04-24:
The print method now clears the screen after executing. This is a breaking change from the previous behavior. Can we revert to leaving the screen as is?

https://github.com/lancaster-university/microbit-dal/releases/tag/v2.2.0-rc4

finneyj, 2020-04-24:
unintentional - apologies @pelikhan 

fixed in https://github.com/lancaster-university/microbit-dal/commit/673228f89a3fbe86eedd48b8d64eaea4bc2a937c

v2.2.0-rc5

pelikhan, 2020-04-25:
All bugs are features by now :)

commit 0fd4a801 (2020-04-21): Correctly initialise animation timer for printChar()  Fix #463