lancaster-university/microbit-dal #468: Bluetooth mesh support

weeg, 2020-08-24:
Is there any plan to support BT Mesh networking?

bluetooth-mdw, 2020-08-25:
FYI Bluetooth mesh just about works on a micro:bit using the Zephyr RTOS. There are some compromises (provisioning and config tend to have to be hard-coded) because micro:bit has so little memory. You can see a demo of Bluetooth mesh on micro:bit here:

https://youtu.be/J0DMsH5YVw4?t=1396 

Not sure we'll ever see it in the DAL, unfortunately but I have no involvement in that decision.

weeg, 2020-08-25:
This demo is very impressing, is there any sample code available that I could use as a starting point?

bluetooth-mdw, 2020-08-25:
Thanks :-)

Yes and no. It was all available in the "Bluetooth Mesh Developer Study Guide", an educational resource for developers from Bluetooth SIG. But in the last release, micro:bit was dropped and replaced with the Nordic Thingy as it has a superior Bluetooth module and more memory. All the same educational points are covered though, both in theory and in practice.

See https://www.bluetooth.com/bluetooth-resources/bluetooth-mesh-developer-study-guide/ 

The Zephyr codebase includes a sample directory with a few mesh examples in it too:

https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/bluetooth 

Cheers

Martin

weeg, 2020-08-25:
Thanks Martin, I will look into these!