lancaster-university/microbit-dal #470: Using the Accelerometer independently of MicroBit uBit;

JamesBrooksEUREKA, 2020-09-19:
Hi All, 
this is a problem I was having earlier in the year but put it to one side for a while. It still persists though and I would be really grateful if there is a solution as with this capability my microbit project isn't viable.

I am wanting to use the accelerometer independently of Micro uBit.

I have no problems using the display independently. Like this.
`#include "MicroBit.h"

MicroBitDisplay display;

int main()
{
    while(1)
        display.scroll(":)");
}`

However if I try and use the accelerometer in the same way, like this.

`#include "MicroBit.h"

MicroBitI2C i2c = MicroBitI2C(I2C_SDA0, I2C_SCL0);
MicroBitAccelerometer accelerometer = MicroBitAccelerometer(i2c);
MicroBitDisplay display;

int main()
{
    while(1)
        display.scroll(accelerometer.getX());
}`

I spent a good few days messing around with it to see if I was doing something wrong and eventually contacted Mark Williams. Who confirmed that it did indeed thrown an error, don't think he was too sure why and directed me here. This was in June of this year.

If anybody has a solution to this, and hopefully a working example. That doesn't use the full MicroBit uBit. I would be eternally grateful as it is a project for work.

Kind Regards

James

JamesBrooksEUREKA, 2020-09-19:
I am using VScode and used the Yotta setup guide. as far as I know, I have the most up to date components.

martinwork, 2020-09-20:
The accelerometer constructors changed around May 2018 to allow for detection of hardware variants.

This works:

```
#include "MicroBit.h"

MicroBitI2C i2c(I2C_SDA0, I2C_SCL0);
MicroBitAccelerometer &accelerometer = MicroBitAccelerometer::autoDetect(i2c);
MicroBitDisplay display;

int main()
{
    while(1)
        display.scroll(accelerometer.getX());
}

```

JamesBrooksEUREKA, 2020-09-20:
Thanks, Martin. not sure why I didn't figure that out at the time. I tried various combinations of stuff but didn't get it quite right. Thanks a lot for the help. I'll post my project when I've finished so other people can have a look.

JamesEBrooks, 2020-10-23:
This is me again but not on my work account. Just thought I'd post the HEX of what I've done so far. Its based on bits from 'The Nature of Code' which I translated into c++. It uses a sort of mini flocking, separation with gravity and mass - each has random mass so reacts differently -. It works on a 50x50 grid that is then turned into a 5x5 grid. The accelerometer lets you move round a sort of  blob that has a bit of a life of its own.  Anyway have  look if your interested.

[https://github.com/JamesEBrooks/MicroBit](url)

JamesEBrooks, 2020-10-23:
I'll post the project as well if anybody is interested, need a bit of tidying up first.

isFakeAccount, 2020-12-13:
Accelerometer always returns 0 for me. I am not sure why.