lancaster-university/microbit-dal #471: Fix right scrolling of an image bug!

arduino12, 2020-09-28:
**Hi,**

I want to be able to scroll an image from left to right,
But apparently- it only works in the web simulator- see my demo of the bug [**here**](https://makecode.microbit.org/_c4FYDbEYvXAi).

I found that the cause of this is [**here**](https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MicroBitDisplay.cpp#L408).

To my thinking - the quick-fix will be to rename `image.shiftLeft(int16_t n)` into `image.shiftHorizontal(int offset)` -
So it can accept a negative value and shift the image to the right as well.

But a better fix will be to rename it into `shift(enum direction)` or `shift(int x_offset, int y_offset)` -
So we can shift an image in all directions!

**Thanks!**

martinwork, 2020-09-30:
How about this? https://github.com/lancaster-university/microbit-dal/pull/427

arduino12, 2020-09-30:
Oh sorry I missed your PR @martinwork - it could solve my issue !

But sadly it seems that nobody here want's to merge it after reading the issue #351 from 2018 :(

Shame that people had to re-implemet this in the higher layer like [**this**](https://github.com/makecode-extensions/ScrollText).

My goal is to add Hebrew font for my students - I wanted to do it in this DAL lower layer -
But it looks like I better do it myself in the .ts files so I don't have to wait 1+ years for a PR approvers.
Anyway I will close this issue now and will write my scrolling code using the .ts files. :)

closed 2020-09-30 by arduino12