lancaster-university/microbit-dal #479: Error in pin pull state on a V2 microbit

whaleygeek, 2021-03-16:
The V2 micro:bit using 'digital read pin' appears to have an incorrect pull state.

On a V1 microbit, reading from a digital pin that is unconnected returns low, on a V2 micro:bit, reading from a digital pin that is unconnected returns high.

I had to add a 'set pull' in the code to make this work on a V2 micro:bit

[image]

makecode.microbit.org version:  3.0.67
Microsoft MakeCode version:  6.0.39
microbit runtime version:  v2.2.0-rc6

cc @microbit-carlos

finneyj, 2021-03-16:
Hi @whaleygeek! Long time no see!

We're tracking V2 tickets over here: https://github.com/lancaster-university/codal-microbit-v2

You're quite right - this was defaulting to "no pull". It should already be fixed codal-microbit-v2#v0.2.25, which was tagged yesterday. It should be making its way into the editors shortly. 

Tracked on issue here:
https://github.com/lancaster-university/codal-microbit-v2/issues/17

microbit-carlos, 2021-03-19:
@whaleygeek this fix has been released to this version of MakeCode: https://makecode.microbit.org/stable
Would you be able to give this a quick test to confirm this has been resolved?

whaleygeek, 2021-03-19:
Hi @microbit-carlos I just tried this on my V2 (remembering to remove the set-pull line from the on start!) and it does indeed work correctly now.

Thanks very much for this. I will let Bill know on Twitter too, as he was the end-user that discovered this.

microbit-carlos, 2021-03-19:
Fantastic, thanks Joe for the fix, David for testing and Bill from the report and all the testing he's done as well!