lancaster-university/microbit-dal #481: Just Works pairing selected but Passkey used and no sign of encryption on link

bluetooth-mdw, 2021-09-22:
This may be a makecode issue or might be a DAL issue. I don't have time to look into this further so I'm hoping that by flagging the issue here, someone connected with micro:bit will look into this.

I just created this simply makecode project:

```
bluetooth.onBluetoothConnected(function () {
    basic.showString("C")
})
bluetooth.onBluetoothDisconnected(function () {
    basic.showString("D")
    basic.pause(5000)
    basic.clearScreen()
})
bluetooth.startAccelerometerService()
```

and in project settings selected Just Works pairing. However when I pair, I get prompted for a 6 digit passkey. This should not happen when using Just Works pairing. Furthermore, after pairing when I connect to the device from a phone using the nRF Connect app and enable accelerometer data notifications, using a protocol analyser I don't see encryption starting on the link and ATT PDU payloads are in clear. There should be some link layer PDUs exchanged to start encryption and the ATT data exchange should be encrypted.

Looks to me like security isn't working.

[image]

martinwork, 2021-09-22:
Thanks @bluetooth-mdw There does seem to be a problem here. I'm not sure what it is at the moment, but I think it's probably in MakeCode. I will create an issue for MakeCode.

One thing to beware... I think WebUSB partial flashing doesn't overwrite the bond data in micro:bit.

Briefly... I didn't see any passkey requested. I did see micro:bit behaving as if security wasn't working. Right now I can't repeat that problem. I'm wondering if there's some caching such that forgetting the pairing doesn't immediately take full effect.

Stranger... After the other tests below, I just selected passkey pairing in MakeCode live and iOS still didn't require a passkey. Having removed the pairing from the iPad, nRF can still connect. Android nRF Connect can see it in a scan (which it shouldn't) but won't connect, saying not bonded, then bonds without a passkey. When using https://makecode.microbit.org/v2, passkey works as expected.

Other details... 
On my Windows machine, I created a new project in MakeCode live, added the BLE extension, pasted in your code, and checked the Settings. Just works is the default, so I didn't change anything. 

I USB full-flashed (by saving to the MICROBIT drive) that to my micro:bit V1. Having removed all pairings from my iPad, the iOS micro:bit app paired up without a passkey and could see accelerometer data, so all seemed OK.

Next, I removed all pairings from my iPad and USB full-flashed from Windows MakeCode live again. Now nRFConnect could see the micro:bit when I refreshed its scan and connect to it. Without a pairing, it shouldn't see a micro:bit that isn't in pairing mode. I got the same results with a micro:bit V2.

When I tried the same experiment built on https://makecode.microbit.org/v2, nRF Connect behaved correctly. It couldn't see the un-paired micro:bit unless it was in pairing mode.

Now I can't repeat the problem with MakeCode live.

bluetooth-mdw, 2021-09-22:
Hi @martinwork 

FYI I didn't use WebUSB to flash. I downloaded the hex file and copied it to the usual USB mass storage drive on my Windows laptop.

Sounds like something odd is going on but that maybe my initial problem report doesn't quite capture whatever nuances apply. Wish I could do more but I'm just too busy unfortunately.

I'll watch this thread and will comment if I think I have something useful to contribute.

Good luck!