lancaster-university/microbit-dal #482: Microbit v2 Compiler

TDanielI, 2021-09-29:
Hello,
I have a question about the installation of the offline development tools (yotta). I followed the instructions given and couldn't manage to do the last step (pip install -U yotta). The error it is showing is "ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly". I have tried to follow the steps given in https://stackoverflow.com/questions/59441794/error-could-not-build-wheels-for-cryptography-which-use-pep-517-and-cannot-be-i to solve it, but it doesn't seem to be working. Could you help me with this or are there any other online development alternatives for microbit v2?

Thank you!

carlosperate, 2021-09-29:
This is an issue with Yotta, which normally should be reported in https://github.com/ARMmbed/yotta, but as the tools is deprecated you are unlikely to get much help over there.

Normally there should be cryptography wheels available for most platforms. What Operating System are you using and what Python version?

This should force pip to look for wheels instead of try to build from source:

```
pip install "cryptography>=2.8,<3" --only-binary ":all:"
```

If that doesn't work you can try to install and old enough version (still compatible with yotta) that was still using C instead of Rust, which should be easier to build from source:

```
pip install "cryptography==2.8"
```

carlosperate, 2021-09-29:
> Could you help me with this or are there any other online development alternatives for microbit v2?

Actually, if you are developing for V2 you should be using this instead, which does not use Yotta: https://github.com/lancaster-university/microbit-v2-samples/

TDanielI, 2021-09-29:
Thank you for the help, I am using Windows 10 64-bit and Python 3.9.7
I tried to do the first command, but I got this error:
ERROR: Could not find a version that satisfies the requirement cryptography<3,>=2.8 (from versions: 3.1, 3.1.1, 3.2, 3.2.1, 3.3, 3.3.1, 3.3.2, 3.4, 3.4.1, 3.4.2, 3.4.3, 3.4.4, 3.4.5, 3.4.
6, 3.4.7, 3.4.8)
ERROR: No matching distribution found for cryptography<3,>=2.8

I can try to install the cryptography independently by using pip install --upgrade cryptography if that's helpful, also upon installing cryptography==2.8, I received the same error as the first one.

As for the last link, do I need to uninstall anything? and should I create another virtual environment for that case?

carlosperate, 2021-09-29:
Right, to be able to use older crytography wheels you will have to use an older Python version (I've used 3.6 for example).

As for using the V2 samples you don't need to uninstall anything, you can use the same venv if you like 👍

TDanielI, 2021-09-29:
What about the gcc-arm-none-eabi, I currently have version 5.4. Do I need to uninstall that to install the new version 10.3? Would there be any problems caused by having both of them?

carlosperate, 2021-09-29:
No problems to have both, just make sure the newer version is in your PATH when building the V2 samples.
Right now there are issues with 10.3, so I'd install an older one instead.
https://github.com/lancaster-university/microbit-v2-samples/issues/33

TDanielI, 2021-09-29:
Which version would you recommend? Is it the v10?

carlosperate, 2021-09-29:
10.2 should be fine

TDanielI, 2021-09-29:
I have a question about the building section of the GitHub you gave:
Building

    Clone this repository
    In the root of this repository, type python build.py
    The hex file will be built MICROBIT.HEX and placed in the root folder.

What does point 2 mean, should I type it on the python console?
Also, just to confirm my understanding about the developing section, is it true that I can compile any file with .h .c and .cpp format by placing the files on the source file?

carlosperate, 2021-09-29:
Hi @TDanielI,

I think at this point it might be better to open an issue in https://github.com/lancaster-university/microbit-v2-samples/ that way other people having similar issues can find the answer to these questions.

TDanielI, 2021-09-29:
Thank you for your help!

closed 2021-09-29 by TDanielI