lancaster-university/microbit-dal #484: Is offline development dead?

stalexwang, 2021-11-18:
I cloned microbit-examples repo, execute 'yt build', then I got the error message:
`error: Can't pickle <function Pack.runScript at 0x10ef25e50>: it's not the same object as yotta.lib.pack.Pack.runScript`

The official yotta project on GitHub is marked as deprecated.
So, does anyone have any idea about the future of microbit-dal?

It's sad the mbed 2.0 version used in DAL was already obsolete even before micro:bit was shipped.
And now yotta is gone.

martinwork, 2021-11-18:
Could that error be related to your local setup? Is that a Python error?
I see there was a recent update to microbit-samples: https://github.com/lancaster-university/microbit-samples/commit/a97ba70394cef088000576a9a6ffd00439e63fff
Maybe it's necessary to grab the latest microbit-samples, or make the same change locally?

stalexwang, 2021-11-18:
Thanks, apparently some fellow developers also ran into this issue.

carlosperate, 2021-12-28:
I think I had the same issue a while ago, but only with Python 3.8+, so I installed yotta with Python3.6 and that worked.

Alternatively if you are familiar with Docker you could use this container image that has all the tooling setup: https://github.com/carlosperate/docker-microbit-toolchain