lancaster-university/microbit-dal #486: triple-tap-reset to enter pairing mode

jaustin, 2023-01-30:
in CODAL we've implemented a really nice feature that lets a user enter pairing mode by triple-tapping the reset button.

https://github.com/lancaster-university/codal-microbit-v2/tree/triple-reset-to-pair

For us to move to this as the primary pairing mechanism we'll need to ensure it's in microbit-dal too.

@finneyj @JohnVidler could you please port this to microbit-dal before we make the next DAL release?

whaleygeek, 2023-02-05:
@jaustin See this, for some motivation to get this done!!!

https://twitter.com/jerseyitguy/status/1622205053640622080

microbit-carlos, 2023-06-29:
We working on this ticket we can take the opportunity and remove some start up delays as well: 
https://github.com/lancaster-university/microbit/blob/1d1baa561f82ab0d16d1916b0026955c67d95e74/source/MicroBit.cpp#L133

martinwork, 2023-07-03:
See https://github.com/lancaster-university/codal-microbit-v2/issues/348

martinwork, 2023-08-14:
Created https://github.com/lancaster-university/microbit/pull/20 with animation delay reduced form 50ms to 20ms.

@microbit-carlos Would you like me also to remove the 100ms delay?
https://github.com/lancaster-university/microbit/blob/1d1baa561f82ab0d16d1916b0026955c67d95e74/source/MicroBit.cpp#L133
I don't see any evidence as to why it was added, but it seems to have always been there.

carlosperate, 2023-08-14:
@finneyj when you were looking at startup timings for the tripple-press-to-pair in V1 DAL, did you notice the delay's from Martin last comment? And do you know if they are safe to remove? (or worth removing?)

microbit-carlos, 2024-04-16:
Any solutions we have thought about to tackle this feature are too invasive or weird, so we've decided not to implement triple reset to pair mode in V1.

closed 2024-04-16 by microbit-carlos