lancaster-university/microbit-dal #487: Precise stepper motor control, precise timing (timer interrupt)?

lafar6502, 2023-02-06:
Hi, i decided to ask the question here as i'm not aware of any low-level micro:bit forum. And sorry about my ignorance, im a newbie to micro:bit trying to understand the programming environments.
So, i would like to approach stepper motor control with the stepstick-like hardware (drv8825, for example).
The idea is to provide two input signals to the stepper driver - step and dir. Step pulse causes the stepper motor to make a step, and dir signal controls the rotation direction.
So, in order to implement fluent and precise motion you need to be able to control the step pulse - its frequency and number of pulses, and be able to adjust these precisely. 
First of all, are you aware if anyone has done that, and with what approach? Maybe the problem is solved already for microbit?
My idea would be to use programmable timer interrupt - divide the motion profile into segments with constant velocity (constant step frequency) and generate the pulses with timer. After completing each segment the timer would be reprogrammed to different frequency, for specified number of pulses, until end of segment is reached.
Is it possible to program hardware timers via CODAL, and if so, are there any bindings for pxt/makecode blocks?
The other possible approach would be to use PWM generator - but here we only have control over the frequency, but no control over the number of pulses generated. Or, maybe i'm wrong and we can control it somehow?
Thanks and best regards
RG

martinwork, 2023-02-13:
Hi @lafar6502 Create a MakeCode project and search the Extensions for "stepper". There are a few MakeCode extensions that deal with stepper motors. Click the Learn more... links to find the source code on GitHub.

As I understand it, the stepper motors themselves don't demand precise or rapid pulses.

This is the repo for issues relating to DAL and micro:bit V1. For CODAL and micro:bit V2 there is https://github.com/lancaster-university/codal-microbit-v2 For MakeCode questions, there is https://forum.makecode.com/. For MakeCode bug reports, feature requests, etc. there is https://github.com/microsoft/pxt-microbit.

lafar6502, 2023-02-16:
Thanks Martin. At the moment i'm a bit lost trying to identify what are the building blocks of the whole MakeCode for Micro:bit infrastructure and how they all interact. But from what i understood, the DAL/CODAL is a firmware that provides the runtime for MakeCode programs, and it uses various resources (timers, interrupts, ...) for that purpose - therefore they're not available for the blocks programs. What is left is 'soft' timers (delays) and direct i/o manipulation which is not guaranteed to be very accurate. But maybe it's accurate enough, i just need to experiment.

closed 2023-02-16 by lafar6502