lancaster-university/microbit-dal #494: Is the id value of the gesture event 27 or 13?

zfangqijun, 2024-01-11:
Excuse me, is the id value of the gesture event 27 or 13? I used Bluetooth to listen for events and found that 13 was valid, but 27 was invalid.

Value 13 from https://github.com/microsoft/pxt-microbit/blob/ade1055f9878be355bf6f5a1ac59005895f7fdbd/libs/core/dal.d.ts#L967

Value 27 from https://github.com/lancaster-university/microbit-dal/blob/602153e9199c28c08fd2561ce7b43bf64e9e7394/inc/core/MicroBitComponent.h#L63

martinwork, 2024-01-11:
In MakeCode it's 13 for both V1 and V2. Here's a test: https://makecode.microbit.org/_J6rD1zLUsRWu

The latest version of this repo is on the js-event-semantics branch, not master.

The js-event-semantics is 36 commits ahead of master
https://github.com/lancaster-university/microbit-dal/tree/js-event-semantics

**Details...**

MakeCode uses the latest tag in the microbit repo (follow link in MakeCode About box)
https://github.com/lancaster-university/microbit/releases/tag/v2.2.0-rc6
which points to the latest tag in this repo
https://github.com/lancaster-university/microbit-dal/releases/tag/v2.2.0-rc6

That tag points to this commit on the js-event-semantics branch https://github.com/lancaster-university/microbit-dal/commit/2443bebacee0eee8d57d032a3d9617c87f50232e

From there, use "Browse files" to find the ID_GESTURE.

The numbers were aligned with V2 CODAL
https://github.com/lancaster-university/microbit-dal/blame/2443bebacee0eee8d57d032a3d9617c87f50232e/inc/core/MicroBitComponent.h#L44

Here it is in CODAL:
https://github.com/lancaster-university/codal-core/blob/63f017fdcd12aca769264c36fe47e6071a40e502/inc/core/CodalComponent.h#L45

zfangqijun, 2024-01-15:
@martinwork 🫡Thank you.

closed 2024-01-15 by zfangqijun