lancaster-university/microbit-dal #496: UART over Serial.

Beat-YT, 2024-11-30:
Currently we can only use UART over Bluetooth, but I want to use it over serial pins.

the python docs have specifications for UART, as seen here: https://microbit-micropython.readthedocs.io/en/v1.0.1/uart.html

but it seems like this is not supported in the c++ environment. Could support for UART be added?

martinwork, 2024-12-01:
I had the same problem looking for the serial module in Python!
https://lancaster-university.github.io/microbit-docs/ubit/serial/

Beat-YT, 2024-12-01:
@martinwork I will try to implement it myself using the source code from bbc micropython.
https://github.com/bbcmicrobit/micropython/blob/v1.0.1/source/microbit/microbituart.cpp

Beat-YT, 2024-12-01:
@remay 
Looks like I can just redirect the serial line to the pins if I understand correctly?

martinwork, 2024-12-01:
Yes
https://lancaster-university.github.io/microbit-docs/ubit/serial/#redirect

martinwork, 2024-12-01:
@Beat-YT On a proper computer now, so typing is easier... In case it helps...

**For micro:bit V1, start here:** https://github.com/lancaster-university/microbit-samples
main repo for issues: https://github.com/lancaster-university/microbit-dal
dependencies: https://github.com/lancaster-university/microbit-dal/blob/master/module.json#L17
In fact, microbit-dal is a dependency of https://github.com/lancaster-university/microbit

The V1 serial API is in this file
https://github.com/lancaster-university/microbit-dal/blob/master/inc/drivers/MicroBitSerial.h#L436

**For micro:bit V2, start here:** https://github.com/lancaster-university/microbit-v2-samples
main repo for issues: https://github.com/lancaster-university/codal-microbit-v2
dependencies: https://github.com/lancaster-university/codal-microbit-v2/blob/master/target.json#L57

The V2 serial API is in this file
https://github.com/lancaster-university/codal-core/blob/master/inc/driver-models/Serial.h#L380

A great way to find clues is to search previous issues, for example:
https://github.com/lancaster-university/microbit-dal/issues?q=is%3Aissue+uart
https://github.com/lancaster-university/codal-microbit-v2/issues?q=is%3Aissue+uart
https://github.com/microsoft/pxt-microbit/issues?q=is%3Aissue%20uart

Beat-YT, 2024-12-02:
thank you! @martinwork

closed 2024-12-02 by Beat-YT