lancaster-university/microbit-dal #497: How do you build this in 2025?

poopman97, 2025-02-09:
I'm really stumped here. I have changed my CMakeList.txt and changed all of the yotta references I have downloaded ble, ble-nrf51822, nrf51-sdk, mbed-classic and added them to the CMakeList but whatever I do when I try to build the library in visual studio always the files MicroBitConfig.h, MicroBitUtil.h, CoordinateSystem.h, device.h, MicroBitAccelerometer.h, MicroBitCompass.h, MicroBitFile.h, FXOS8700.h, LSM303Magnetometer.h and ble/UUID.h are missing.

Can someone please tell me how do I build this on windows or just send me an already built library so I can just code for the microbit.
Thanks in advanced.

finneyj, 2025-02-10:
Hi @poopman97. 

This repo is built for the v1 version of micro:bit. Is that what you're trying to compile? We have a different repo for the commonly available v2 series of devices, which you can find over here: https://github.com/lancaster-university/microbit-v2-samples

poopman97, 2025-02-10:
I have a V1. 5

finneyj, 2025-02-10:
Ah, ok. Then you're in the right place!

Are you trying to build using the microbit-samples repository?

finneyj, 2025-02-10:
This is the normal starting position for a C based build: https://github.com/lancaster-university/microbit-samples.

If you clone that and following the README in that repo, do you still have problems?

finneyj, 2025-02-10:
Microbit.h is contained in the microbit repo (https://github.com/lancaster-university/microbit).

We factored in out to help ease integration into platforms that just want to use microbit-dal as a big bag of components...

Yes, yotta was a nice tool created by Arm that we used at launch, but has proven a challenge since it was deprecated in 2020. We only use it on out legacy v1 builds now. There aren't a lot of people trying to create new build environments for v1 devices (they're mostly on v2 now)...

Have you tried a manual install of yotta? As I recall it was quite self contained?

poopman97, 2025-02-10:
I managed to install yotta using pip with python 3.8  and downgrading markupsafe to 2.0.1.
But trying to build the samples library gives me this error:
`yt build`
`info: get versions for bbc-microbit-classic-gcc`
`info: download bbc-microbit-classic-gcc@v0.2.2 from GitHub lancaster-university/yotta-target-bbc-microbit-classic-gcc`
`info: get versions for mbed-gcc`
`info: download mbed-gcc@v0.1.3 from GitHub lancaster-university/target-mbed-gcc`
`warning: The build target has not been set, so the default (bbc-microbit-classic-gcc,https://github.com/lancaster-university/yotta-target-bbc-microbit-classic-gcc) is being used.`
`You can use yotta target <targetname> to set the build target. See http://docs.yottabuild.org/tutorial/targets.html for more information on using targets.`
`info: get versions for microbit`
`info: download microbit@v2.1.1 from GitHub lancaster-university/microbit`
`info: get versions for microbit-dal`
`info: download microbit-dal@v2.1.1 from GitHub lancaster-university/microbit-dal`
`info: get versions for mbed-classic`
`info: download mbed-classic@microbit_hfclk+mb6 from GitHub lancaster-university/mbed-classic`
`info: get versions for ble`
`info: download ble@v2.5.0+mb3 from GitHub lancaster-university/BLE_API`
`info: get versions for ble-nrf51822`
`info: download ble-nrf51822@v2.5.0+mb7 from GitHub lancaster-university/nRF51822`
`info: get versions for nrf51-sdk`
`warning: github rate limit for anonymous requests exceeded: you must log in
error: Expecting value: line 1 column 1 (char 0)`

finneyj, 2025-02-11:
hmmm.. "warning: github rate limit for anonymous requests exceeded" looks like you might need to share your github credentials with yotta there? perhaps it won't rate limit if you aren't anonymous?

poopman97, 2025-02-11:
[yt build.txt](https://github.com/user-attachments/files/18751396/yt.build.txt)
[yt install.txt](https://github.com/user-attachments/files/18751401/yt.install.txt)

It's closer now.
I don't see how I can change the nrf51-sdk or ble-nrf51822 to be compatible with each other.
I have ninja and cmake installed.

finneyj, 2025-02-11:
Hmm... still builds for me. I'm using CMake 3.30.2 and arm-none-eabi-gcc-10.2.1.exe:

[image]

poopman97, 2025-02-11:
I'm using arm-none-eabi-gcc-14.2 rel1.
I'll try later to downgrade to your version.
Can you also send me your ninja version?

finneyj, 2025-02-11:
ninja v1.11.1

(very onomatopoeic!)

finneyj, 2025-02-11:
Something could be missing an include.... it may be easier to rewind the version of your tools that it would be to fix this (and other) issues that pop up....

[image]

poopman97, 2025-02-12:
[yt build.txt](https://github.com/user-attachments/files/18767663/yt.build.txt)
Made it farther now, but It fails on the last package.
gcc is 10.3.1 since I would have to compile 10.2.1. It should work though 
[https://github.com/lancaster-university/microbit-dal/issues/493](url)
ninja is 1.11.1.

poopman97, 2025-02-12:
[yt build.txt](https://github.com/user-attachments/files/18769785/yt.build.txt)
It's a different error now but still halts at the last package

finneyj, 2025-02-12:
That looks like the final step post-compilation - it's trying to generate the hex file from the compiled binaries.

It's quite a long command line. I wonder if you're hitting the fabled maximum line length problem on windows? Maybe try a cut and paste of the final line it's stating a run it. See what you get?

remay, 2025-02-12:
Looks like the quoting is broken in that final command.   Too long since I
used this to have any suggestions on how to fix it.

finneyj, 2025-02-12:
Hey @remay! Long time no see!

poopman97, 2025-02-12:
The command mentions srec_cat which could be referring to srecord is installed and I've added the path in sysdm.cpl, environment variables to: C:\Program Files\srecord\bin.
I don't know how to test if it's working.

remay, 2025-02-12:
The error says "The syntax of the command is incorrect" - which I believe
means the shell can't parse the command.

The command looks like:

C:\WINDOWS\system32\cmd.exe /C "cd . &&
C:\PROGRA~2\GNUARM~1\102021~1.10\bin\AR10B2~1.EXE ... -T"..."   .....

I'd wager that those quotes around the -T linker option to gcc need
escaping (and the same with some quotes later in that commandline)    But
I've no recollection as to why that might be a problem for you ot how to
fix it.  I have some recollection that paths with spaces are problematic.

What shell are you running yotta from?

poopman97, 2025-02-12:
Interchangeably cmd and powershell, both generate the same result.

finneyj, 2025-02-12:
Hmm..  I always used gitbash on windows, but these days prefer a full unix environment hosted via WSL2...

Gitbash might be an easy test?

poopman97, 2025-02-12:
[git bash.txt](https://github.com/user-attachments/files/18773464/git.bash.txt)
I think it's the same error.
I really would not install bloated wsl and I don't have an other use for it except this. I have another machine for linux testing.

finneyj, 2025-02-12:
Yes, I agree, it looks like the same error.

It's a mammoth command line though. Perhaps you can break it up into smaller chunks for testing? You should be able to separate it at each '&&' position. That might help us to identify which part of this is causing the problem?